IanHarvey/bluepy #172: NoneType bug in waitForNotifications

LucaCerina, 2017-02-20:
Hi,
I have installed BluePy with pip3 and during a test with long range communication to a notifying peripheral I encountered this error:

`    if p.waitForNotifications(30.0):
  File "/usr/local/lib/python3.5/dist-packages/bluepy/btle.py", line 473, in waitForNotifications
    resp = self._getResp(['ntfy','ind'], timeout)
  File "/usr/local/lib/python3.5/dist-packages/bluepy/btle.py", line 334, in _getResp
    resp = self._waitResp(wantType + ['ntfy', 'ind'], timeout)
  File "/usr/local/lib/python3.5/dist-packages/bluepy/btle.py", line 267, in _waitResp
    if self._helper.poll() is not None:
AttributeError: 'NoneType' object has no attribute 'poll'`

probably because the data packet was corrupted and not received correctly

lilingdan, 2017-03-16:
have you resolved it? My program comes with the problem sometimes , and i dont know how to solve it?

ptcoregon, 2017-03-17:
I'm also having this issue when I try to stop a scan and then start again

IanHarvey, 2017-06-19:
Can you post some code which shows this problem?

IzidorM, 2018-03-19:
Hi,
i also have this exception showing up frequently. It happens if the device get disconnected and after that the waitForNotification is called...

caiki, 2019-01-03:
> Can you post some code which shows this problem?
Here it is.
  File "/home/importbluepy.py", line 72, in readHR
    self.hrm.waitForNotifications(2.0)
  File "/home/charliecaiki/.local/lib/python3.5/site-packages/bluepy/btle.py", line 560, in waitForNotifications
    resp = self._getResp(['ntfy','ind'], timeout)
  File "/home/charliecaiki/.local/lib/python3.5/site-packages/bluepy/btle.py", line 407, in _getResp
    resp = self._waitResp(wantType + ['ntfy', 'ind'], timeout)
  File "/home/charliecaiki/.local/lib/python3.5/site-packages/bluepy/btle.py", line 338, in _waitResp
    if self._helper.poll() is not None:
AttributeError: 'NoneType' object has no attribute 'poll'

```
> 	def readHR(self):
72 		self.hrm.waitForNotifications(2.0)
> 		hrkey = time.time()-self.t0
> 		hrvalue = self.hrm.delegate.bpm[-1]
> 		return (hrkey, hrvalue)
```
In python 2 works well, but in python 3.5 show this error.

Watschl, 2019-07-09:
Hello guys,

has this been resolved? I'm also running in this exception..

tbecker8625, 2020-11-16:
Has there been any change in the status of this problem?  In my judgment this looks like a bug in btle.

rcslProject, 2020-12-09:
in my case, the NoneType bug sometimes happen when I disconnect bluetooth device. my solution is as follow
`btle.Peripheral(self.address)._stopHelper()
 btle.Peripheral(self.address).disconnect()`

xdds1997C, 2021-06-09:
in my case, due to low version of Bluez
so, I upgrade Bluez from 5.50 to 5.58
I use the following command to upgrade Bluez:

sudo apt-get update

sudo apt-get upgrade

sudo apt-get install libusb-dev libdbus-1-dev libglib2.0-dev

sudo apt-get install libudev-dev libical-dev libreadline-dev

then, download Bluez5.58:

wget http://www.kernel.org/pub/linux/bluetooth/bluez-5.58.tar.xz

tar xvf bluez-5.58.tar.xz

cd bluez-5.58

./configure

make

make install