IanHarvey/bluepy #173: Does the test code for btle.py represent the current api?

timcoote, 2017-02-21:
While trying to get my head around the bluetooth le model, I tried to run the code at the bottom of btle.py, but automating the identification of a mac address (as these keep changing).

However, at line 722:
```
    for svc in conn.services:
```
The type of svc is <class bluepy.btle.UUID>, although I think that it is supposed to be <bluepy.btle.Service>. Thus the line 724:
```
            for ch in svc.getCharacteristics():
```
fails with:
```
AttributeError: UUID instance has no attribute 'getCharacteristics'
```
If svc is used to get the service with conn.getServiceByUUID, the code seems to work:
```
        for svcuuid in conn.services:
            svc = conn.getServiceByUUID (svcuuid)
```

Have I misunderstood how this should work, or is there a bug in this test code?

timcoote, 2017-02-21:
I forgot to mention that after the creation of conn = Peripheral (addr, type)  - line 720 - I had to add a call to conn.getServices() to get the conn.services property populated.

I'm using  bluez-5.43-1.fc25.i686, on a stock fedora 25 build with an external btle dongle. btle devices are simulated with PunchThrough's BLE tool on an iPad

IanHarvey, 2017-06-19:
I'm not sure how this can happen with the latest code. In this, the `services` property returns `self._serviceMap.values()`, which is set up by `discoverServices()`, which creates `Service` objects for each value in the dictionary. 

Could you try again with latest code and see if the problem still persists? If so, can you add
```python
        print (repr(conn._serviceMap))
```
after the `for svc in conn.services:` line, and tell me what happens?