IanHarvey/bluepy #177: what's the correct usage of function:"waiting for notification"?

lilingdan, 2017-03-17:
I want to realize the function: send cmd to ble dev while listen notification from it,but when programming,i referred the notification example from the doc [http://ianharvey.github.io/bluepy-doc/notifications.html](url)

```
import btle

class MyDelegate(btle.DefaultDelegate):
    def __init__(self, params):
        btle.DefaultDelegate.__init__(self)
        # ... initialise here

    def handleNotification(self, cHandle, data):
        # ... perhaps check cHandle
        # ... process 'data'

# Initialisation  -------

p = btle.Peripheral( address )
p.setDelegate( MyDelegate(params) )

# Setup to turn notifications on, e.g.
#   svc = p.getServiceByUUID( service_uuid )
#   ch = svc.getCharacteristics( char_uuid )[0]
#   ch.write( setup_data )

# Main loop --------

while True:
    if p.waitForNotifications(1.0):
        # handleNotification() was called
        continue

    print "Waiting..."
    # Perhaps do something else here
```
    to send and listen notification at the same time, i put listen notification task(code: While True:....) in a thread. But it always comes with Exception(bluepy.btle.BTLEException: Unexpected response (wr)) then break down, i know it warns me that  function "writeCharacteristic" is failed ,but In fact it succeeds and i get notification that i need.

    so i try to put "waitForNotifications" out of while, in another word, i just call it once in my program, amazing, i can get notification normally instead of breaking down after receiving first notification.

    then i tried delete "waitForNotifications", amazing, my program works fine!!!
    so i just wonder what's the correct usage of function:"waiting for notification", i hope you can help me, thx.

dbarclay-wg, 2017-03-29:
By any chance, are you calling "waitForNotifications" in a background thread?
I solved this in my code by putting the Characteristic.write() in the same thread as Peripheral.waitForNotifications() so that they do not conflict.

mone741120, 2017-10-18:
Execute me, I have similar problem.
If I want to connect multiple peripherals, and receive notification of them.
Also maybe I need to use different thread for each device to handle the notification.
In this case, How can I use waitForNotifications function on different peripherals?

Snapsion, 2018-02-05:
@dbarclay-wg Thanks for the tip. Also waitForNotifications also conflicts with .read() method.