IanHarvey/bluepy #182: bluepy.btle.BTLEException: Unexpected response (stat)

danielunderwood, 2017-04-07:
I'm getting the following error:

```
Traceback (most recent call last):
  File "blep.py", line 97, in <module>
    main()
  File "blep.py", line 46, in main
    scanned_devices = [d.addr for d in scanner.scan(timeout=POLL_INTERVAL)]
  File "/usr/local/lib/python2.7/dist-packages/bluepy/btle.py", line 631, in scan
    self.start()
  File "/usr/local/lib/python2.7/dist-packages/bluepy/btle.py", line 579, in start
    self._mgmtCmd("scan")
  File "/usr/local/lib/python2.7/dist-packages/bluepy/btle.py", line 236, in _mgmtCmd
    rsp = self._waitResp('mgmt')
  File "/usr/local/lib/python2.7/dist-packages/bluepy/btle.py", line 301, in _waitResp
    raise BTLEException(BTLEException.INTERNAL_ERROR, "Unexpected response (%s)" % respType)
bluepy.btle.BTLEException: Unexpected response (stat)
```

It looks like a status command is being returned from something, but `Scanner.scan()` is the only call that I ever make. I was able to resolve the issue with a reboot, but it would be nice to figure out a fix that doesn't involve rebooting.

I'm using python 2.7.9 on a Pi Zero W if that matters.

EinSoldiatGott, 2017-05-18:
Did you find the problem? Same error here
BTLEException occurred. 3, 'Unexpected response (stat)'

danielunderwood, 2017-05-18:
I never found what the actual problem was as I never saw the problem again after a reboot. I've done several updates since then as well, so I'm not really sure what it was. I'd try to update bluepy, your system packages, and do a reboot to see if that fixes it.

EinSoldiatGott, 2017-05-18:
I'm doing a scan every 20sec to find new devices, I get this error very often, fortunately y have a try/catch to handle it, but I'd be better not having errors.

I',m using bluez 5.29 in raspberry 3 with bluepy 1.10

intrepiddigital, 2018-01-03:
Hi I have successfully used the following code in Python 2.7.12 and have no problems getting data from a BLE device. You will recognise the code.

```
from bluepy.btle import Scanner, DefaultDelegate
class ScanDelegate(DefaultDelegate):
    def __init__(self):
        DefaultDelegate.__init__(self)

    def handleDiscovery(self, dev, isNewDev, isNewData):
          device_name = ""          
while True:
    try:
        scanner = Scanner().withDelegate(ScanDelegate())
        devices = scanner.scan(.1,0)
        for dev in devices:
                for (adtype, desc, value) in dev.getScanData():
                    if desc == "Complete Local Name":
                        device_name = value[:7]
                        print(datetime.datetime.now())
                        if device_name == "Connect":
                            Post_json(value,dev.addr)
    except KeyboardInterrupt:
        print('Cnt + C detected')
        sys.exit()
    except RuntimeError as e:
        print('Problem with the Bluetooth adapter : {}'.format(e))
```

However if I run the code under Python 3.5.2 I get 

bluepy.btle.BTLEException: Unexpected response (stat)

It appears to be dependent on the scan period. If i leave at .scan(.1,0) I get the error above in python 3. If i increase to .scan(2) it is fairly safe.
.scan(.1.0) always runs for ever in Python 2.7.12 with no errors.

Does anyone have any ideas?

danielunderwood, 2018-01-03:
That's interesting. Perhaps I saw the error when adjusting the scan interval. I'll be moving a few projects over to python 3 in the next few weeks, so I'll see if I can reproduce this error effectively.

intrepiddigital, 2018-01-03:
I will be interested to see if you have a worse problem under Python 3. The reason i need to scan so frequently is because the device changes its name when a button is pressed and the response needs to be immediate. ....but as I said, I have no issues under 2,7.

intrepiddigital, 2018-01-03:
Maybe I should do the correct thing and leave my full traceback.

`Traceback (most recent call last):
  File "scanfordevice_integrated_3.5.py", line 28, in <module>
    devices = scanner.scan(.1,0)
  File "/usr/local/lib/python3.5/dist-packages/bluepy/btle.py", line 681, in scan
    self.stop()
  File "/usr/local/lib/python3.5/dist-packages/bluepy/btle.py", line 630, in stop
    self._mgmtCmd(self._cmd()+"end")
  File "/usr/local/lib/python3.5/dist-packages/bluepy/btle.py", line 272, in _mgmtCmd
    rsp = self._waitResp('mgmt')
  File "/usr/local/lib/python3.5/dist-packages/bluepy/btle.py", line 337, in _waitResp
    raise BTLEException(BTLEException.INTERNAL_ERROR, "Unexpected response (%s)" % respType)
bluepy.btle.BTLEException: Unexpected response (stat)
`

ketanbhaid, 2018-02-22:
I am getting the same error does  anyone got the solution to this problem

akelson, 2018-05-31:
I noticed this error when I was accidentally attempting to scan in two different processes at the same time.

EinSoldiatGott, 2018-07-10:
I run multiple scans on different hci indexes, I haven't seen a correlation between the error and the simultaneous scanning. But having multiple interfaces.