IanHarvey/bluepy #184: does bluepy can run as an peripheral and central?

DanielTimLee, 2017-04-14:
I mean in BLE 4.x, GAP Profile has some profiles, mostly the one who really provides data (Peripheral), and the other one who collects data (Cetnral).

Peripheral Advertises itself, and wait for Central's connection request.

I think Bluepy just have only Central role. I can't find `bluepy` doc that says it advertises itself to other Central Devices. 

Am I right?... (I'm not sure about this)
If there are no Peripheral in `bluepy`, then will does Peripheral role is it planned to be developed?

Endres, 2017-05-11:
I would be also interested in such a feature.

jan-xyz, 2017-05-11:
As far as I know it is using bluez to do all the Bluetooth stuff and last time I checked bluez didn't support any peripheral mode. I might be wrong and it might be available in newer versions.

Endres, 2017-05-11:
indeed, bluez does support at least advertising, as stated in various raspberry pi iBeacon tutorials. It is possible using the `hciconfig` and `hcitool` command line:

```
hciconfig hci0 leadv 3
hcitool -i hci0 cmd 0x08 0x0008 1e 02 01 1a 1a ff fe ff 02 15 [...] (zero-filled 32 bytes in total)
```

So this should be at least doable using raw HCI commands. Maybe also with bluepy. But another solution could be to simply use pybluez, which supports raw HCI commands and also has a couple of examples which use that. But this needs some low-level engineering just to be able to advertise as peripheral. I don't know how extensive this would get when even applying the role of a GATT server and allow for connections from Centrals.

ukBaz, 2017-05-11:
FYI:
You might be interested to take a look at this talk about the capabilities of BlueZ
https://youtu.be/tclS9arLFzk
The slides are available at:
http://events.linuxfoundation.org/sites/events/files/slides/Bluetooth%20on%20Modern%20Linux_0.pdf

BLE Advertising is available under the "Experimental" flag in the DBus API.
Rather than using hcitool the preferred way is with the btmgmt command line tool

IanHarvey, 2017-08-29:
So I've tried and largely failed to get BlueZ's peripheral support to do anything useful, on a couple of   occasions. I'm not, I think, going to use this approach.

I do have some experimental code which talks straight to a Bluetooth socket from Python and gives very simple Peripheral behaviour. If anyone wants a project to play with, and not a finished library,  I'll put it up on GitHub -  please ask.

Thanks
Ian

AnDrOiD-73, 2017-09-04:
**IanHarvey**, I will be very grateful if you put it up on GitHub )

ukBaz, 2017-09-04:
@IanHarvey and @TheBubbleworks: it might be worth the pair of you exchanging ideas on using Bluetooth sockets for peripherals.
There are some examples from @wkeenan at:
https://github.com/TheBubbleworks/python-hcipy

WayneKeenan, 2017-09-04:
Please do publish what you've got, cheers.

IanHarvey, 2017-09-10:
Here you go: https://github.com/IanHarvey/bluepy.device

There's not a proper API for using it yet, and there is plenty of missing functionality, but it might be enough to get something simple going.

Enjoy...
IH

closed 2018-09-22 by IanHarvey