IanHarvey/bluepy #185: Where to start?

navid9675, 2017-04-19:
Hi this is not an issue just asking for guidance ... I'm new in ble stuff and I was wondering where should I start leaning ? is there any tutorial . 
Thank you

IanHarvey, 2017-05-07:
Hi - the place I started was with a Texas Instruments Sensortag - http://www.ti.com/ww/en/wireless_connectivity/sensortag/ - which is a reasonably well-documented little device which lets you read lots of sensors via BLE. They have getting-started guides and a smartphone app which lets you check it's all working OK without having to write any software.

Once you're familiar with that bluepy should basically just work. There's a utility program `sensortag` included which will read Sensortag data, and gives a bit of an example of how to do Bluetooth programming.

Hope that helps,
Ian

IanHarvey, 2017-07-07:
I've written a bit of a guide to getting started, using a sensortag and a Raspberry Pi, at http://www.elinux.org/RPi_Bluetooth_LE .

If you have any feedback, let me know.

Thanks
Ian

closed 2018-02-21 by navid9675