IanHarvey/bluepy #186: Could you create a new release?  The last release was about a year ago, and there have been several enhancements made since.

chriswhsu, 2017-04-21:


rtyley, 2017-05-03:
Can I offer 0.02 bitcoin (~$30) towards making a new release of bluepy, incorporating the various [updates](https://github.com/IanHarvey/bluepy/compare/v/1.0.5...feb4690f5b12d) to master since v1.0.5?

Specifically I'm interested in the descriptor stuff added in #89 (because, I think, I just learned that it's necessary to get notifications working).

Thanks for all the hard work you've done so far - you make a difference to a lot of people.

IanHarvey, 2017-05-07:
Umm, time flies when you're not looking.

I've had a bit of a tidy-up of outstanding pull requests and updated bluepy to 1.1.0, which should now be available on PyPi.

I'm aware that the docs at http://ianharvey.github.io/bluepy-doc/ are also behind the times, and that the bluez code version (5.29) is very old now and could do with an update.

(I'm very flattered; if you really wanted to make a contribution, I'm in turn indebted to both Mozilla.org and the Wikimedia Foundation for what they provide for free, so a donation to either would be good).

Thanks
Ian

rtyley, 2017-05-07:
> I've had a bit of a tidy-up of outstanding pull requests and updated bluepy to 1.1.0, which should now be available on PyPi.

Brilliant - I can confirm that I've successfully installed this updated bluepy version on a fresh Raspbian Jessie install, using [these instructions from Adafruit](https://learn.adafruit.com/install-bluez-on-the-raspberry-pi/installation) for the BlueZ install:

```
$ sudo apt-get install -y libusb-dev libdbus-1-dev libglib2.0-dev libudev-dev libical-dev libreadline-dev

$ wget http://www.kernel.org/pub/linux/bluetooth/bluez-5.45.tar.xz
$ tar xvf bluez-5.45.tar.xz 
$ cd bluez-5.45/

$ ./configure
$ make
$ sudo make install

$ sudo systemctl daemon-reload
$ systemctl status bluetooth

$ pip install bluepy
```

> I'm aware that the docs at http://ianharvey.github.io/bluepy-doc/ are also behind the times

It _would_ be good to have a few basic examples front and centre, with the correct imports (I think it's `from bluepy import btle`, is that right?)

> I'm in turn indebted to both Mozilla.org and the Wikimedia Foundation for what they provide for free, so a donation to either would be good

Sure, I've donated $20 each to [Mozilla](https://www.coinbase.com/orders/4141a002a593ebf7fa187a34f0cea871/receipt) and [Wikimedia](https://www.coinbase.com/orders/ba7f7f64b304a1f48fc1b110429ef979/receipt) - thanks again!

IanHarvey, 2017-08-29:
I wrote a bit of a tutorial up at http://www.elinux.org/RPi_Bluetooth_LE which I hope will take people through the basics of getting it set up. I'm guessing absolute beginners will need to get basic command-line communication working before trying to write programs. It's on a public wiki so feel free to amend it...

Cheers
Ian

closed 2017-08-29 by IanHarvey