IanHarvey/bluepy #190: Permissions Error

raykamp, 2017-05-17:
Hey Ian,  I'm getting the following error when running the "blescan" utility. It's a similar issue to https://github.com/IanHarvey/bluepy/issues/94, but I am using sudo.

```sh
$ sudo /usr/local/bin/blescan
Scanning for devices...
Traceback (most recent call last):
  File "/usr/local/bin/blescan", line 9, in <module>
    load_entry_point('bluepy==1.1.0', 'console_scripts', 'blescan')()
  File "/usr/local/lib/python2.7/dist-packages/bluepy/blescan.py", line 122, in main
    devices = scanner.scan(arg.timeout)
  File "/usr/local/lib/python2.7/dist-packages/bluepy/btle.py", line 674, in scan
    self.start()
  File "/usr/local/lib/python2.7/dist-packages/bluepy/btle.py", line 612, in start
    self._mgmtCmd("le on")
  File "/usr/local/lib/python2.7/dist-packages/bluepy/btle.py", line 276, in _mgmtCmd
    "Failed to execute mgmt cmd '%s'" % (cmd))
bluepy.btle.BTLEException: Failed to execute mgmt cmd 'le on'
```
I'm running Ubuntu 16.04.2 on an x86-64 machine.  
**Is there something I'm doing wrong here?** 

I did try using this method to give "capabilities" to the bluepy-helper program. It didn't work. 
https://unix.stackexchange.com/questions/96106/bluetooth-le-scan-as-non-root

raykamp, 2017-05-17:
If I run bluepy-helper directly and with sudo, I am able to execute 'le on' and 'scan'

```sh
$ sudo ./bluepy-helper 1
le on
rsp=$mgmt code=$success
scan
rsp=$mgmt code=$success
rsp=$stat state=$scan dst='(null) mtu=h0 sec='low
rsp=$scan addr=b43221686C114 type=h2 rssi=h3B flag=h0 d=b02010613FF4C000C0E08D97FFED6EEE0B8CB242ECAE0AB
rsp=$scan addr=b54E18ED171AF type=h2 rssi=h3B flag=h0 d=b02010613FF4C000C0E00965954BF988565E47E8EBAD0D3
rsp=$scan addr=bACBC329E198E type=h1 rssi=h3B flag=h0 d=b02010607FF4C0010020B00
rsp=$scan addr=bA47733FC1DED type=h1 rssi=h43 flag=h0 d=b0201060303A0FE1016A0FE03FA8FCA960EF6612020203EFF06096F43617374
```

closed 2017-05-18 by raykamp

henridwyer, 2017-06-27:
@raykamp I actually got this to work using the link you provided on a Fedora server:

sudo setcap 'cap_net_raw,cap_net_admin+eip' PATH/TO/LIB/python3.5/site-packages/bluepy/bluepy-helper

Make sure it's the right version of the executable.

djnotus, 2017-11-23:
The issue is that this line in blte.py in the Scanner Class has the interface hard coded to 1. Run hciconfig and see what interfaces you have, and change the 0 to the number of the interface you are trying to use.

def __init__(self, iface=0):

Cesarec88, 2018-05-07:
Hello, I've implemented the solution proposed,
still...

Traceback (most recent call last):
  File "main.py", line 48, in <module>
    run()
  File "main.py", line 27, in run
    scan_dev = scan_mng.filter_device_by_addr(scan_mng.scan_devices(), mac_addressto_filter)
  File "/home/pi/master/scan.py", line 48, in scan_devices
    return self.__scanner.scan(scan_time)
  File "/home/pi/.local/lib/python3.5/site-packages/bluepy/btle.py", line 679, in scan
    self.start(passive=passive)
  File "/home/pi/.local/lib/python3.5/site-packages/bluepy/btle.py", line 617, in start
    self._mgmtCmd("le on")
  File "/home/pi/.local/lib/python3.5/site-packages/bluepy/btle.py", line 276, in _mgmtCmd
    "Failed to execute mgmt cmd '%s'" % (cmd))
bluepy.btle.BTLEException: Failed to execute mgmt cmd 'le on'

what can I do?

henry-stann, 2018-06-02:
Test comment

henry-stann, 2018-06-02:
Hey Cesarec88,

Can you quickly confirm that you are finding bluetooth devices at all? One of the below should work:

pi@raspberrypi:~ $ sudo /usr/local/bin/blescan -i=0
pi@raspberrypi:~ $ sudo /usr/local/bin/blescan -i=1
pi@raspberrypi:~ $ sudo /usr/local/bin/blescan -i=2

If none of them do, you can try:

$cd ~
$cd  /usr/local/lib/python3.5/dist-packages/bluepy
$ sudo ./bluepy-helper 1
le on
rsp=$mgmt code=$success
scan
rsp=$mgmt code=$success
# (ctrl-C to escape, and try sudo ./bluepy-helper 0 and sudo ./bluepy-helper 2)

If none of those work, you can try:
pi@raspberrypi:~ $ sudo  /usr/local/lib/python3.5/dist-packages/bluepy/bluepy-helper 0
le on
rsp=$mgmt code=$success
scan
rsp=$mgmt code=$success
# (ctrl-C to escape, and try 0 and 2)

My number is actually different with the two different ways to scan. I'm just trying to scan in as many different ways as possible to make sure your linux can read anything. 

Can I confirm, also, that you implemented djnotus' comment by using:
pi@raspberrypi:~ $ sudo nano /usr/local/lib/python3.5/dist-packages/bluepy/btle.py
Use Ctrl+_ (ctrl+shift+'minus-sign') to go to line 605:
class Scanner(BluepyHelper):
    def __init__(self,iface=0):
And changed that 0 to a 1, if that is your trouble? It was not mine. 

Am I correct to assume that you are running a python script? The 'main.py' in your traceback tells me something's amiss. Can I confirm that you are indeed running python from terminal using sudo? 

I use Ian Harvey's code, http://ianharvey.github.io/bluepy-doc/scanner.html . My test file is as follows:

pi@raspberrypi:~ $ cat Desktop/sample.py
from bluepy.btle import Scanner, DefaultDelegate

class ScanDelegate(DefaultDelegate):
    def __init__(self):
        DefaultDelegate.__init__(self)

    def handleDiscovery(self, dev, isNewDev, isNewData):
        if isNewDev:
            print "Discovered device", dev.addr
        elif isNewData:
            print "Received new data from", dev.addr

scanner = Scanner().withDelegate(ScanDelegate())
devices = scanner.scan(10.0)

for dev in devices:
    print "Device %s (%s), RSSI=%d dB" % (dev.addr, dev.addrType, dev.rssi)
    for (adtype, desc, value) in dev.getScanData():
        print "  %s = %s" % (desc, value)

I run it by: 
pi@raspberrypi:~ $ sudo python Desktop/sample.py

Sorry if the incredibly detailed walk-thru seems like it is babying you. Often, I get tangled in linux by omitting very small details.

fleutot, 2021-08-10:
@henridwyer:
> @raykamp I actually got this to work without needing sudo, using the link you provided on a Fedora server:
> 
> sudo setcap 'cap_net_raw,cap_net_admin+eip' PATH/TO/LIB/python3.5/site-packages/bluepy/bluepy-helper
> 
> Make sure it's the right version of the executable.

Great! I want to keep bluepy in my projects local virtualenv. I added capabilities to that file anyway and it works. 

But I wonder about delivering to others. I can't see any other way than asking users to run an install script that creates the venv, pip installs the requirements.txt, then run the command above and prompt for superuser password. Do you?