IanHarvey/bluepy #191: Graceful helper shutdown after receiving KeyboardInterrupt

danskarda, 2017-05-17:
I handle KeyboardInterrupt in main thread to gracefully handle application shutdown.

When I press Control-C, helper application is terminated and BluepyHelper cannot parse response (see stack below).

I found [StackOverflow answer](http://stackoverflow.com/questions/5045771/python-how-to-prevent-subprocesses-from-receiving-ctrl-c-control-c-sigint) which suggests to call Popen with preexec_fn parameter and ignore SIGINT.

Should I modify btle.py? Are there any other hidden impacts of masking the interrupt? What do you think?

```
  File "/usr/local/lib/python3.5/dist-packages/bluepy/btle.py", line 376, in disconnect
    self._getResp('stat')
  File "/usr/local/lib/python3.5/dist-packages/bluepy/btle.py", line 334, in _getResp
    resp = self._waitResp(wantType + ['ntfy', 'ind'], timeout)
  File "/usr/local/lib/python3.5/dist-packages/bluepy/btle.py", line 281, in _waitResp
    resp = BluepyHelper.parseResp(rv)
  File "/usr/local/lib/python3.5/dist-packages/bluepy/btle.py", line 246, in parseResp
    (tag, tval) = item.split('=')
ValueError: not enough values to unpack (expected 2, got 1)
```

IanHarvey, 2017-06-19:
The fact that you get `ValueError` when the helper exits like this is a bug. I've just pushed change f2d014a which detects this condition and raises `BTLEException(BTLEException.INTERNAL_ERROR)` if bluepy-helper gets killed.

Does this fix your problem - i.e. are you happy to catch this exception and exit gracefully? I don't really want to ignore Ctrl-C in bluepy-helper, because it's potentially useful behaviour, and there are other things (e.g. `kill -9`) which can't be readily ignored.

Thanks
Ian

mobilinkd, 2018-03-30:
This is bad behavior.   bluepy-helper must ignore Ctrl-C because otherwise it prevents applications from shutting down gracefully when the user presses Ctrl-C.  For example, for bonded devices it may be important to disable notification before the device disconnects (it is for me) because notification state for bonded devices must be preserved by the peripheral.  Having notification enabled on a newly connected device may cause unexpected behavior.  The only way for me to make a properly conforming application is to catch KeyboardInterrupt, disable notifications, then exit cleanly.

Assuming the developer wants the background process killed by Ctrl-C is dangerous.  If you really want to ensure that the background process goes away, register an atexit handler to explicitly kill it.

mobilinkd, 2018-03-30:
Pull request submitted for this issue.

https://github.com/IanHarvey/bluepy/pull/264

closed 2018-11-11 by IanHarvey