IanHarvey/bluepy #193: Writing greater than 20 bytes

elchung, 2017-05-22:
Hi, 

is it possible to write more than 20 bytes of data with a single characteristic.write() command? For some reason, when I try sending data that is greater than 20 bytes, only the first 20 bytes are sent over. I know that 20 bytes is the limit to sending data in a single "chunk" for BLE applications, but does bluepy backend take care of chopping large bytes of data into 20 byte chunks and sending the "write execute" after all the data is sent? 

When I try writing more than 20 bytes with a single write() command, it appears that only the first 20 bytes are sent, and nothing else is sent.

Sorry if this is a simple question, as I am quite new at this. Thanks for any help!

jardous, 2018-08-01:
This is limitation of BT 4.0

anilgurses, 2019-09-02:
BT 5.0 support writing more than 20 bytes. However, the library is not up to date. I think the library is cutting data at some point. I looked for that in the files. I didn't find anything up to now. I will inform you when I find

anilgurses, 2019-09-02:
Hi,
I found a temporary solution. You can use bigger MTU size. Before everything, you should be aware of your hardware limits. Just
```
p = btle.Peripheral("00:00:00:00:00:00")
p.setMTU(200) # byte size you need
```

Jaftem, 2019-09-04:
@anilgurses

Thanks for the suggestion.

This doesn't seem to work on my end. Central is a RPi3 and peripheral is an nRF51 device. I've gotten a higher MTU to work with Noble, so I believe this is still possibly a bluepy issue?

brett-thornhill, 2021-07-08:
I am seeing this issue with reading. 

If I have the line `p.setMTU(200)  ` (Doesn't matter how large the number), the message will be cut off.

If I take out that line, then I can read the entire message using `characteristic.read()` but it still is cut off using notifications.

Is there a way to increase the amount of data that is received via `handleNotification(self, cHandle, data)`?