IanHarvey/bluepy #194: waitForNotifications() without using a while but a NotificationCenter.

stretchsenselouisgermain, 2017-05-29:
Hey guys,

I've been using Bluepy since 3 months for my application and it comes more and more concrete.

What I would like now is not to open a new thread using a while in the function waitForNotifications() because it is not really the best for my application.

My idea is to use an observer and notifier as the NS NotificationCenter from swift programming language, I tried it, it works but I still need to loop at a moment and in iOS programming that's internal, I guess their notification use isn't implemented the same way.

I put you the link from the NotificationCenter function/class I'm using, if you need to have a look.

https://gist.github.com/lantiga/288267

If someone have an idea, I'll be extremely glad, it will help me a lot!

Thanks,

Louis

ukBaz, 2017-05-30:
Hello,

I am not familiar with Swift programming and the NS NotificationCenter but after reading [this article](https://developer.apple.com/reference/foundation/nsnotification) it seemed very similar to [DBus](https://pythonhosted.org/txdbus/dbus_overview.html) which can be used with BlueZ to get notifications from Bluetoothd. DBus does require the use of an event loop and adopting an async style of programming.

stretchsenselouisgermain, 2017-05-30:
Hey ukBaz,

I had a look on all of this, I'm not sure that it can be use or I may not understand how it works.
Because I'm mostly using Python and the bluepy API in my program and the idea is to use the waitForNotification() function not doing a thread with the while.

The idea is more : having a notification center like the one in Switch language and the one I have linked on the post ahead that when a observer detects a notification from the postNotification function we jump to the function which is link to the observer by breaking the program.

I don't know if it's possible.

Cheers,

Louis

IanHarvey, 2017-06-19:
On iOS / Mac OS, calling of observers is usually done by an `CFRunLoop` object; the OS automatically runs one of these as part of every application's main thread. There's a reasonably good description at http://bou.io/RunRunLoopRun.html

The nearest equivalent I know of in Python is the `asyncio` module (https://docs.python.org/3/library/asyncio.html) which allows you to write programs in an event-driven style.  It's pretty complicated (and you have to write the whole program in this style) so I didn't want to use it for `bluepy`.

Thanks
Ian

stretchsenselouisgermain, 2017-07-10:
Thanks Ian for giving that library, I have implemented it in my code and also using the bluepy and it works pretty well.

Cheers
Louis.

sneko, 2020-08-10:
Hi @stretchsenselouisgermain ,

It's been a while since you posted that but... question... My whole Python program uses `asyncio` so I can "parallelize" (still in 1 thread) some tasks that would be blocking without this. It works well on the overall program but for the Bluetooth part it's still a burden:

* `asyncio` stays in a "instructions thread" (not a real thread) until it finds an `await`, where it can allow another async task to run... and so on...
* my problem is if I use `bluepy` to scan devices for 10 seconds, it will block 10 seconds onto this instruction and `asyncio` in the meantime won't give resource to another async task

I'm looking for a viable alternative. Did you find it? Wondering if I should consider another BLE library like: https://github.com/hbldh/bleak

Thank you,

thevickypedia, 2020-11-25:
Potentially an irrelevant comment - How do I get it installed on a MacOS?
I even did a `brew install glib` version `2.66.2_1` before trying to install `bluepy`

```
fatal error: 'byteswap.h' file not found
#include` <byteswap.h>
1 error generated.
make: *** [bluepy-helper] Error 1"
  ----------------------------------------
  ERROR: Failed building wheel for bluepy
```

Any insights/help would be much appreciated

carolcabral, 2021-01-13:
Hey @sneko,

Did you make it work? bluepy with asyncio? Or did you change to other library?  If so, which one?

sneko, 2021-01-13:
@carolcabral I switched to https://github.com/hbldh/bleak , it works great for now :)