IanHarvey/bluepy #197: not module named btle

sdy1130, 2017-06-13:
[image]

I installed this library as instructed and I get this error when I run the same code...

BillyNate, 2017-06-13:
''

PrzemoF, 2017-06-14:
pip or local build? Did you try import bluepy.btle ?
Also you can always try to use your distro reporitory to install bluepy.

IanHarvey, 2017-06-19:
If you installed the library with pip, the recommended way is:
```python
from bluepy import btle
```

Does this work for you? Maybe the documentation needs updating to make this clear.

closed 2017-11-13 by IanHarvey

g2kokaro, 2018-06-23:
I am having the same issue. I'm running Ubuntu 18.04 with Python 3.6.5. I've tried installing via pip3 both with and without sudo and installing from source.
[image]

willm1001, 2018-08-17:
If you have the bluepy source directory in the same directory as the python file that is calling bluepy, delete, move, or rename it. I git cloned the bluepy source to my home directory and also had created my python test file in my home directory and it caused this problem. Once I removed the bluepy source directory, I was able to call bluepy without issue.

Further info:
You will need libglib2.0-dev for the glib.h requirement...
```
sudo apt-get install libglib2.0-dev
wget <path to bluepy source tarball, get from the download link above>
tar -xzvf bluepy-1.0.5.tar.gz
cd bluepy-1.0.5
sudo python3 setup.py build
sudo python3 setup.py install
```

jml6969, 2019-07-31:
from bluepy import btle
from bluepy import sensortag (*from bluepy.bluepy import sensortag won't work!*)

if you have been doing the example from Orestis from github (before coming to this example) make sure to change the name of file (sensortag.py) of this example to something else. This can create issue.