IanHarvey/bluepy #199: Failed to install bluepy with all means

fonics, 2017-06-28:
I have python 3.4 installed (64) .

I have tried " pip install bluepy" with no success.
[image]

jan-xyz, 2017-06-29:
It looks like you are trying to use the library on Windows.
The first sentence in the [README](https://github.com/IanHarvey/bluepy/blob/master/README.md) says:
```
Python interface to Bluetooth LE on Linux.
```

fonics, 2017-06-29:
thanks @jan-xyz  sorry for the inconvenience

closed 2017-07-04 by IanHarvey

abdulwajid016, 2018-12-26:
I got this error in installing bluepy
I want to get the raw data from BLE Beacon 
If you guys solve my problem then i will be really greatful to you
[image]