IanHarvey/bluepy #200: difficulties install python2.7

matak, 2017-07-01:
cc1: warnings being treated as errors
./bluez-5.29/src/shared/att.c: In function âbt_att_send_error_rspâ:
./bluez-5.29/src/shared/util.h:131: error: dereferencing pointer â__pâ does break strict-aliasing rules
./bluez-5.29/src/shared/util.h:131: note: initialized from here
make: *** [bluepy-helper] Error 1
make: Leaving directory `/root/bluepy/bluepy'

anybody can help?

lsb_release -d
Description: CentOS release 6.8 (Final)

IanHarvey, 2017-07-04:
This may be caused by your system's `gcc` being too new for the rather old (5.29) version of the BlueZ source code.

Could you tell me what `gcc --version` says, on your system?

Thanks
Ian

IanHarvey, 2017-07-09:
I've made a new release (1.1.1, now available on pypi.org) which should not have this problem, as I've disabled `-Werror`, which is causing the installation to fail. I can't promise, however, that this is the only fix required.

Let me know if this makes a difference.
Thanks
Ian

closed 2017-09-19 by IanHarvey