IanHarvey/bluepy #201: Device getting disconnected when waiting for Notification

ivantot2, 2017-07-06:
Hi all,

I seem to always be disconnected whenever i try to wait for a notification from my device.

I get a bluepy.btle.BTLEException: Device disconnected

here is my sample code

```python
import binascii
import struct
import time
from bluepy.btle import *

class MyDelegate(DefaultDelegate):
    def __init__(self):
        DefaultDelegate.__init__(self)

    def handleNotification(self, cHandle, data):
        print("Data is: {}".format(data))

temp_uuid = UUID("be15bee1-6186-407e-8381-0bd89c4d8df4")

p=Peripheral("e5:27:27:28:13:be","random")

p.setDelegate(MyDelegate())

print("connected")
try:
    ch = p.getCharacteristics(uuid=temp_uuid)[0]
    p.writeCharacteristic(0x000b,b"\x01\x00",withResponse=True)
    print("open")
    p.writeCharacteristic(0x000b,b"\x01\x1a")
    i = 0
    while i < 10:
        if p.waitForNotifications(1.0):
            print("Notification")
            continue
        i+=1
        print("waiting")

finally:
    p.disconnect()
```

i get this result

> connected
> open
> waiting
> waiting
> Traceback (most recent call last):
>   File "sampleReadFromCar.py", line 31, in <module>
>     if p.waitForNotifications(1.0):
>   File "/usr/local/lib/python3.4/dist-packages/bluepy/btle.py", line 516, in waitForNotifications
>     resp = self._getResp(['ntfy','ind'], timeout)
>   File "/usr/local/lib/python3.4/dist-packages/bluepy/btle.py", line 369, in _getResp
>     resp = self._waitResp(wantType + ['ntfy', 'ind'], timeout)
>   File "/usr/local/lib/python3.4/dist-packages/bluepy/btle.py", line 326, in _waitResp
>     raise BTLEException(BTLEException.DISCONNECTED, "Device disconnected")
> bluepy.btle.BTLEException: Device disconnected

Am i doing something wrong?

gandy92, 2017-07-13:
I got notifications working by writing to the characteristics' hande+1, like this:

```
try:
    ch = p.getCharacteristics(uuid=temp_uuid)[0]
    handle = ch.getHandle()
    p.writeCharacteristic(handle+1,b"\x01\x00",withResponse=True)
    print("open")
    # p.writeCharacteristic(0x000b,b"\x01\x1a")
    # here I'd write to some other characteristic that triggers notifications
    i = 0
    while i < 10:
        if p.waitForNotifications(1.0):
            print("Notification")
            continue
        i+=1
        print("waiting")

finally:
    p.disconnect()
```

ivantot2, 2017-07-14:
hi, another question if you have the time,

the code above gets the battery value, and the value is different everytime
the notification gets a message.
it ranges from 0-255(ex. notification 1=[3,27,200,16], notification
2=[3,27,5,16]

can you guide me on how i can get the battery correctly? im looking for a
0-100 value and reliable value not the one that changes every time i
recieve a battery message

Jarvz, 2017-07-21:
If the lists you posted are your data, you just need to format them. Assuming that the list is a set of pairs of uint8's, something like:
```
def handleNotification(self, cHandle, data):
        new_data = []
        val_data = struct.unpack("B"*len(data),data)
        for i in range(0,len(data)-1,2):
            new_data.append(val_data[i]+(val_data[i+1]<<8))
        print(new_data)
```
Would unpack the data and combine pairs of uint8's back into uint16's.

kiranbharwani, 2020-05-26:
Hello,

Trying this out on Raspberry Pi. And getting Device Disconnected error. I know the device it on. 
Any help would be appreciated. 
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: bluepy in /usr/local/lib/python2.7/dist-packages (1.3.0)

Code:
from bluepy import btle

class BPDelegate(btle.DefaultDelegate):
    def __init__(self):
        print("Init")
        btle.DefaultDelegate.__init__(self)

    def handleNotification(self, cHandle, data):
        print("A notification was received: %s" %data)
        print(data.decode("utf-8"))

print("Connecting ...")
p = btle.Peripheral("28:ff:b2:16:48:1f") 
p.withDelegate( BPDelegate() )

services=p.getServices()
for service in services:
   print service

# Setup to turn notifications on, e.g.
svc = p.getServiceByUUID( '00001810-0000-1000-8000-00805f9b34fb' )
ch = svc.getCharacteristics()[0]

p.writeCharacteristic(ch.getHandle()+1, b"\x02\x00", withResponse=True)

while True:
    if p.waitForNotifications(1.0):
        # handleNotification() was called
        continue

    print("Waiting...")
    # Perhaps do something else here_

Output:
Connecting ...
Init
Service <uuid=ecbe3980-c9a2-11e1-b1bd-0002a5d5c51b handleStart=768 handleEnd=930>
Service <uuid=Generic Access handleStart=1 handleEnd=81>
Service <uuid=Generic Attribute handleStart=256 handleEnd=274>
Service <uuid=Battery Service handleStart=1280 handleEnd=1298>
Service <uuid=Blood Pressure handleStart=1024 handleEnd=1057>
Service <uuid=Device Information handleStart=512 handleEnd=641>
Service <uuid=Current Time Service handleStart=1536 handleEnd=1554>
Characteristic <Blood Pressure Measurement>
1041
Traceback (most recent call last):
  File "test1.py", line 35, in <module>
    if p.waitForNotifications(1.0):
  File "/usr/local/lib/python2.7/dist-packages/bluepy/btle.py", line 560, in waitForNotifications
    resp = self._getResp(['ntfy','ind'], timeout)
  File "/usr/local/lib/python2.7/dist-packages/bluepy/btle.py", line 407, in _getResp
    resp = self._waitResp(wantType + ['ntfy', 'ind'], timeout)
  File "/usr/local/lib/python2.7/dist-packages/bluepy/btle.py", line 362, in _waitResp
    raise BTLEDisconnectError("Device disconnected", resp)
bluepy.btle.BTLEDisconnectError: Device disconnected