IanHarvey/bluepy #204: Install issue on raspberry pi zero

LucaCerina, 2017-07-19:
Hi,
I have noticed that the isntallation of bluepy (both from pip or source compilation) does not link well the modules, because:

'from bluepy import btle'

goes well if done in the main folder of bluepy, while it returns "cannot import name btle" anywhere else.

ozeta, 2017-07-20:
I have the same problem

rdvmikko, 2017-07-20:
Copy&paste from one of my source files:
from bluepy.btle import Scanner, DefaultDelegate, Peripheral, BTLEException

ozeta, 2017-07-20:
sorry, i forgot to say that it doesn't work on python3

LucaCerina, 2017-07-20:
the solution of @rdvmikko instead raises a ModuleNotFoundError, but works if the script is executed in the bluepy folder

rdvmikko, 2017-07-20:
Sorry to hear that. I'm stuck with python 2.7, no problems. Have you checked the python paths and env? I think it might not be a bluepy issue, more like python setup issue maybe..?

ozeta, 2017-07-20:
I don't know. as the matter of fact, using python 2.7 I can succesfully import the module, but I'm working in 3.4 . how do I check the python setup?

IanHarvey, 2017-07-20:
Python 3 keeps its packages in a different place (typically `/usr/local/lib/python3.4/dist-packages`) so installing `bluepy` with Python 2's `pip` won't work. (This isn't surprising as much Python 2 code won't work in 3).

On the Pi, Python 3 has a `pip3` which can be installed with:

    sudo apt-get install python3-pip

You can then install Bluepy with:

    sudo pip3 install bluepy

Hope that helps,
Ian

ozeta, 2017-07-20:
I'll let you know tomorrow!

LucaCerina, 2017-07-21:
I have already tried both using pip3 and source build (which I believe should lead to the same results...), both of them not working.

The solution for me was to upgrade BlueZ on the raspi from 5.23 (default in Raspbian Jessie) to 5.29, and now it works. I am guessing that the wrong version causes some unhandled exception during the installation

ozeta, 2017-07-21:
in my case 
`
sudo pip3 install bluepy`

worked just fine! (in my system pip3 was already installed)

IanHarvey, 2017-08-29:
OK, I'll close this issue for now. I don't suppose you still have a log of any error messages from `pip3` ? There have been a few other installation failures reported, and I'm wondering if there is a common reason.

Thanks
Ian

closed 2017-08-29 by IanHarvey

enfrte, 2019-09-05:
Not sure if this helps anyone, but I had a similar issue, and it was because I had created a test script called bluepy.py in my home folder. Deleting/renaming the file solved the problem. It seems like python scans the home folder first for library things.