IanHarvey/bluepy #205: issue running on pi3

gesaleh, 2017-07-24:
Hi I have installed the lib and trying to run any of the examples all fails with the same error
```
Traceback (most recent call last):
  File "/home/pi/Sitesense/ble/ble_bluepy.py", line 535, in <module>
    for d in discover_devices():
  File "/home/pi/Sitesense/ble/ble_bluepy.py", line 518, in discover_devices
    for d in scanner.scan(timeout = 0.1):
  File "/usr/local/lib/python2.7/dist-packages/bluepy/btle.py", line 674, in scan
    self.start()
  File "/usr/local/lib/python2.7/dist-packages/bluepy/btle.py", line 611, in start
    self._startHelper(iface=self.iface)
  File "/usr/local/lib/python2.7/dist-packages/bluepy/btle.py", line 246, in _startHelper
    universal_newlines=True)
  File "/usr/lib/python2.7/subprocess.py", line 710, in __init__
    errread, errwrite)
  File "/usr/lib/python2.7/subprocess.py", line 1335, in _execute_child
    raise child_exception
OSError: [Errno 2] No such file or directory
```
is this a known issue ?
or did I missed something

IanHarvey, 2017-08-29:
_bluepy_ needs an executable called `bluepy-helper` to be compiled from C source code during installation. It should be in the same directory as the main `btle.py` file. The error indicates that this file is missing - either the setup program wasn't run, or it failed to compile `bluepy-helper` properly.

How did you install the library (from source on github, or from PyPI, or elsewhere?) Can you post what messages were printed during installation?

Thanks
Ian

IanHarvey, 2017-11-13:
I'm now closing this as a duplicate of #158. This should (!) be fixed in the 1.1.4 release.

Thanks
Ian

closed 2017-11-13 by IanHarvey