IanHarvey/bluepy #209: Why ignore btle.Peripheral contructor arguments?

henridwyer, 2017-08-06:
This library is great!But I don't understand why the __init__ arguments of 
btle.Peripheral are ignored: everything is set to None
Also, why does btle.Peripheral.connect require an address, and not just use the peripheral address? (Actually this makes sense since the address passed into the constructor is ignored, but if it weren't ...)

If you're interested I can submit a PR to update this behavior.

InvncibiltyCloak, 2017-08-27:
If you follow the code down a little further, self.deviceAddr, self.addrType and self.iface are set to the correct values within self._connect that is called in init.

IanHarvey, 2017-09-19:
These are _default arguments_ which can be omitted when calling the method.

There's a proper explanation here: https://docs.python.org/3/tutorial/controlflow.html#default-argument-values

Cheers
Ian

closed 2017-09-19 by IanHarvey

henridwyer, 2017-09-19:
I guess I should have given an example. It seems to me that this should work:

``` python
peripheral = Peripheral(addr)
peripheral.connect()
```