IanHarvey/bluepy #216: Confirming BLE connection

neal-tommy82, 2017-08-30:
Ian, 

I wonder if you can help... I have Bluepy working on the Raspberry Pi3 connecting to a Thunderboard Sense board. All works well over long periods of time (reporting once every 10 minutes  and uploading to cloud), however sometimes it disconnects by itself. 

Could you give some advice as to how to make the system more reliable : 

- is there a way to confirm if the BLE is still connected? Is there something in the Bluepy library to handle this?
- is there a way to reboot the system if the Raspberry Pi realises there is no activity?

Thanks for your help! 

Neal.

garryblr, 2018-08-02:
Hi, Did you got ans for your post?
Am also in same phase

neal-tommy82, 2018-08-02:
@garryblr - I ended up not using this functionality and seem to have found code which gets the RPi to connect to one specific MAC address if it is broadcasting, otherwise wait. 

Hope this helps.

PrzemoF, 2019-01-21:
https://github.com/IanHarvey/bluepy/pull/334 please test & close if it fixes the issue