IanHarvey/bluepy #218: bluepy.btle.BTLEException: Failed to execute mgmt cmd 'le on'

edouardkleinhans, 2017-09-04:
Hello, 

I'm trying to run you sample code (http://ianharvey.github.io/bluepy-doc/scanner.html#sample-code) on my orangepi with a bluetoot v4 dongle, but it doesn"t work.

root@orangepipcplus:~# lsusb 
Bus 008 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 007 Device 002: ID 0a12:0001 Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode)

root@orangepipcplus:~# hcitool dev
Devices:
	hci0	00:1A:7D:DA:71:13

root@orangepipcplus:~# python test.py 
Traceback (most recent call last):
  File "test.py", line 14, in <module>
    devices = scanner.scan(10.0)
  File "/usr/local/lib/python2.7/dist-packages/bluepy/btle.py", line 679, in scan
    self.start(passive=passive)
  File "/usr/local/lib/python2.7/dist-packages/bluepy/btle.py", line 617, in start
    self._mgmtCmd("le on")
  File "/usr/local/lib/python2.7/dist-packages/bluepy/btle.py", line 276, in _mgmtCmd
    "Failed to execute mgmt cmd '%s'" % (cmd))
bluepy.btle.BTLEException: Failed to execute mgmt cmd 'le on'

Any idea of what's wrong ?

PrzemoF, 2017-09-04:
Did you try with sudo?

edouardkleinhans, 2017-09-04:
all the command are executed as root.

goldfingyr, 2017-09-08:
I think I have the same problem just with the TI Sensortag (running as root - debugging on). Running on a Raspberry Pi with a BLE USB:

 sudo python sensortagcollector.py
Running  /usr/local/lib/python2.7/dist-packages/bluepy/bluepy-helper
Sent:  le on

Got: ''
Got: ''
Got: ''
Got: ''
Got: ''
Got: ''
Sent:  le on

Traceback (most recent call last):
  File "sensortagcollector.py", line 317, in <module>
    devices = scanner.scan(timeout=30.0)
  File "/usr/local/lib/python2.7/dist-packages/bluepy/btle.py", line 674, in scan
    self.start()
  File "/usr/local/lib/python2.7/dist-packages/bluepy/btle.py", line 612, in start
    self._mgmtCmd("le on")
  File "/usr/local/lib/python2.7/dist-packages/bluepy/btle.py", line 271, in _mgmtCmd
    self._writeCmd(cmd + '\n')
  File "/usr/local/lib/python2.7/dist-packages/bluepy/btle.py", line 267, in _writeCmd
    self._helper.stdin.write(cmd)
IOError: [Errno 32] Broken pipe

rmesnard, 2017-09-20:
Same issue for me.
On a Orange PI Zero , i've try 2 different  USB adapter ( CSR 4.0 and Orico )  Both working perfectly with hcitools. 

`Scanning for devices...
Traceback (most recent call last):
  File "/usr/local/bin/blescan", line 9, in <module>
    load_entry_point('bluepy==1.1.2', 'console_scripts', 'blescan')()
  File "/usr/local/lib/python2.7/dist-packages/bluepy/blescan.py", line 122, in main
    devices = scanner.scan(arg.timeout)
  File "/usr/local/lib/python2.7/dist-packages/bluepy/bluepy/btle.py", line 679, in scan
    self.start(passive=passive)
  File "/usr/local/lib/python2.7/dist-packages/bluepy/bluepy/btle.py", line 617, in start
    self._mgmtCmd("le on")
  File "/usr/local/lib/python2.7/dist-packages/bluepy/bluepy/btle.py", line 276, in _mgmtCmd
    "Failed to execute mgmt cmd '%s'" % (cmd))
bluepy.bluepy.btle.BTLEException: Failed to execute mgmt cmd 'le on'
`
I try to rebuild helper with the same results.

Ishma59, 2017-10-18:
same problem on my kali linux 2017  I try running  : 
$sudo bleah t0
BTLE Exception:
Failed to execute mgmt cmd 'le on'

Ishma59, 2017-10-18:
any help thanks in advance

songyawen, 2017-10-19:
I have the same question. (2017kali\bleah ) Thanks for any help!!!!!!

SHA89, 2017-11-15:
What output does bluepy-helper give when you run it directly?
`/usr/local/lib/python2.7/dist-packages/bluepy/bluepy-helper 0
help
le on`

rosshadden, 2017-12-08:
`sudo` makes it work for me. Do we know which group or permissions it requires? I would like to make it run with my user.

SHA89, 2017-12-11:
`sudo setcap 'cap_net_raw,cap_net_admin+eip' bluepy-helper`
This gives everyone permission to execute bluepy-helper.

SHA89, 2018-02-19:
have you tried running bluepy-helper directly and tried to execute "le on" manually?

hasamba, 2018-02-19:
im getting: "bluepy-helper: command not found"
eventough i can see the file and it is executeable

SHA89, 2018-02-20:
I guess you are not so familiar with linux command line.
to run the executable, you must execute it the following way:
`./bluepy-helper 0`
If you forget the "./", the system will assume you try to start a program that is available in your PATH.
With the "./" you give the direct address of your executable.
The "0" defines the interface you want to use. If you only have one bluetooth interface, 0 will be correct,
if you have multiple, you can use 1, ...

hasamba, 2018-02-20:
thanks for your explanation, i learned something new :)
im running now /usr/local/lib/python3.5/dist-packages/bluepy $ ./bluepy-helper 0 help le on
and im not getting any response, just the curser blinking, is that ok?

SHA89, 2018-02-20:
if you type 'help' you will see all possibilities.
As i can see in your error message, your problem already occurs when trying to execute the command
`le on` (to put it in bluetooth low energy mode)
so i suggest you try that command.

hasamba, 2018-02-20:
for some reason "help" also doesnt give me any result...just blinking

SHA89, 2018-02-20:
i'm using it in python 2.7 but i'll set up a 3.5 to test

SHA89, 2018-02-20:
`(venv) ~/python_sandbox/test_bluepy3/venv/lib/python3.5/site-packages/bluepy $sudo ./bluepy-helper 0
help
#help                                           Show this help
#stat                                           Show current status
#quit                                           Exit interactive mode
#conn            [address [address type [interface]]] Connect to a remote device
#disc                                           Disconnect from a remote device
#svcs            [UUID]                         Primary Service Discovery
#incl            [start hnd [end hnd]]          Find Included Services
#char            [start hnd [end hnd [UUID]]]   Characteristics Discovery
#desc            [start hnd] [end hnd]          Characteristics Descriptor Discovery
#rd              <handle>                       Characteristics Value/Descriptor Read by handle
#rdu             <UUID> [start hnd] [end hnd]   Characteristics Value/Descriptor Read by UUID
#wrr             <handle> <new value>           Characteristic Value Write (Write Request)
#wr              <handle> <new value>           Characteristic Value Write (No response)
#secu            [low | medium | high]          Set security level. Default: low
#mtu             <value>                        Exchange MTU for GATT/ATT
#le              [on | off]                     Control LE feature on the controller
#pairable        [on | off]                     Control PAIRABLE feature on the controller
#pair                                           Start pairing with the device
#unpair                                         Start unpairing with the device
#scan                                           Start scan
#scanend                                        Force scan end
#pasv                                           Start passive scan
#pasvend                                        Force passive scan end
rsp=$stat state=$disc mtu=h0 sec='low
`

It seems to work on my computer. I also have a blinking cursor, but if i type help, i do get this help message.
And have you tried other commands? Do you see "help" appearing on your blinking cursor as you type? don't forget to press enter.

SHA89, 2018-02-20:
Just to be sure, you have to start the program with
`sudo ./bluepy-helper 0`
and after it's started, you need to type 'help' or 'le on'

flatsiedatsie, 2018-02-20:
Same error, but when doing multiple calls in quick succession.
Python 3.4 on a Pi Zero W with Dietpi Stretch.

2018-02-20 17:23:54.673 Error: (Mi Flower Mate) ----> Line 55 in /home/pi/domoticz/plugins/mi_flower_mate/miflora/backends/bluepy.py, function connect
2018-02-20 17:23:54.673 Error: (Mi Flower Mate) ----> Line 353 in /usr/local/lib/python3.4/dist-packages/bluepy/btle.py, function __init__
2018-02-20 17:23:54.674 Error: (Mi Flower Mate) ----> Line 402 in /usr/local/lib/python3.4/dist-packages/bluepy/btle.py, function _connect

(working on a plugin that uses this library: https://github.com/flatsiedatsie/Mi_Flower_mate_plugin)

shumingli708, 2018-02-20:
I typed 'le on' on bluepy-helper, and I got 'rsp=$mgmt code=$protoerr'.

hasamba, 2018-02-21:
now im able to run and getting same message: rsp=$mgmt code=$protoerr

SHA89, 2018-02-21:
i also get the protoerr if i'm not running as superuser (sudo ./bluepy-helper 0).
1. So first you need to check if you have the problem as sudo-user
`sudo ./bluepy-helper 0`
`le on`
2. If it works with sudo, and you want to be able to run it as a normal user, you need to set capabilities on the executable. try:
`sudo setcap 'cap_net_raw,cap_net_admin+eip' bluepy-helper`
(make sure u are in the directory with the bluepy-helper executable)
I don't know exactly what this does, but it allows me to run bluepy-helper without root privileges (sudo)
3. Now you can run it as a normal user with 
`./bluepy-helper 0`

If you have problems, please post the commands you used and the output.
FIY i'm running it on ubuntu 16.04 and tested it with python 3.5

hasamba, 2018-02-21:
i ran it before with sudo.

`sudo ./bluepy-helper 0
le on
rsp=$mgmt code=$protoerr
`

im on mint (ubuntu 16.04), python 3.5.2

SHA89, 2018-02-21:
In that case i don't have an explanation, my solution was the capabilities..
Maybe have a try with gatttool to see if you can set up a connection, and hciconfig to check if your bluetooth interfaces are up and running

hasamba, 2018-02-21:
thanks again for all your help

flatsiedatsie, 2018-02-26:
Here the issue still persists if there are multiple calls in a row:

https://www.domoticz.com/forum/viewtopic.php?f=65&t=22281

jamestouri, 2018-02-27:
Hi, was anyone able to solve this problem? I have been stuck on it for a while and I still get the same error
bluepy.btle.BTLEException: Failed to execute mgmt cmd 'le on'

mohithkalyan, 2018-06-26:
Failed to execute mgmt cmd 'le on'.

warpdriv, 2019-01-11:
Ran @PrzemoF 's post and got rsp=$mgmtcode=$mgmterrestat=h11emsg='Invalid Index

on 16.04, py 3.7

Help us Obiwon Kanobi! You're our only hope!

lokeshh, 2019-01-17:
Try `sudo hciconfig hci0 down && sudo hciconfig hci0 up`.

roshini-radha, 2019-03-29:
> I'm still testing, but this works [EDITED]:
> 
> 1. Start from one terminal:
>    `sudo /usr/local/lib/python3.6/site-packages/bluepy/bluepy-helper 0`
>    or wherever you have bluepy-helper
> 2. Type "le on" and you should get this:
>    `rsp=$mgmt code=$success`
> 3. Do whatever you need to do from another terminal.

I tried the same and I got:
rsp=$mgmtcode=$mgmterrestat=hBemsg='Rejected

What should I do next?

Tasm-Devil, 2019-05-04:
Hi.
The Error still exists.
With sudo it works:
`
me@mypc:~$ sudo ./.local/lib/python3.6/site-packages/bluepy/bluepy-helper 0
# bluepy-helper.c version 1.3.0 built at 20:40:35 on May  3 2019
le on
rsp=$mgmtcode=$success
`

But without sudo it wont:
`
me@mypc:~$ sudo setcap 'cap_net_raw,cap_net_admin+eip' ./.local/lib/python3.6/site-packages/bluepy/bluepy-helper
me@mypc:~$ ./.local/lib/python3.6/site-packages/bluepy/bluepy-helper 0
# bluepy-helper.c version 1.3.0 built at 20:40:35 on May  3 2019
le on
rsp=$mgmtcode=$mgmterrestat=h14emsg='Permission Denied
`

Are there any plans to fix it. I want to use [**bt-mqtt-gateway**](https://github.com/zewelor/bt-mqtt-gateway) to get a connection to an nrf52840 running circuitpython.

doronator, 2019-07-09:
Note that when running python scripts with sudo,
your normal user and the super-user ivoked by sudo, 
may have a different default python location.

When using sudo with running a python script you may need to specify the full system path to the version of python you inteend to run

dima72, 2020-08-08:
I'm running on Rasbian PC.
`sudo ./bluepy-helper 0`
```
le on
rsp=$mgmtcode=$mgmterrestat=h11emsg='Invalid Index
```

please suggest what might be wrong

fphammerle, 2021-05-22:
Re: https://github.com/IanHarvey/bluepy/issues/218#issuecomment-350680433
> `sudo setcap 'cap_net_raw,cap_net_admin+eip' bluepy-helper`
> This gives everyone permission to execute bluepy-helper.

`cap_net_admin,cap_net_raw+ep` without `i` ("inheritable") is sufficient:
```sh
$ sudo setcap cap_net_raw,cap_net_admin+ep bluepy-helper
$ getcap bluepy-helper
bluepy-helper = cap_net_admin,cap_net_raw+ep
$ bluepy-helper
# bluepy-helper.c version 1.3.0 built at 06:27:58 on Jul 16 2019
le on
rsp=$mgmtcode=$success
$ uname -r
5.10.17-v7l+
```

EricRongkai, 2021-05-27:
> I'm running on Rasbian PC.
> `sudo ./bluepy-helper 0`
> 
> and getting
> 
> ```
> le on
> rsp=$mgmtcode=$mgmterrestat=h11emsg='Invalid Index
> ```
> 
> please suggest what might be wrong?

Firstly, make sure that your BLE device is existing, using following command

`$ hciconfig`

If the device is existing, it will like this

`hci0:	Type: Primary  Bus: USB
	        BD Address: 00:1A:7D:DA:71:11  ACL MTU: 310:10  SCO MTU: 64:8
	        UP RUNNING 
	        RX bytes:883983 acl:0 sco:0 events:23598 errors:0
	        TX bytes:6053 acl:0 sco:0 commands:160 errors:0
`

Your error is because that the BLE device is not activated. You can use this command to activate it.

`$ sudo hciconfig [hci0] up`

santobedi, 2021-11-02:
Hello @EricRongkai,

`$ hciconfig` gave me
 ```
Command 'hciconfig' not found, but can be installed with:

sudo snap install bluez  # version 5.48-3, or
sudo apt  install bluez  # version 5.53-0ubuntu3.3

```
I installed` sudo apt  install bluez  # version 5.53-0ubuntu3.3` and ran `$ sudo hciconfig [hci0] up`, however, I get `Can't get device info: No such device`. Same thing happened with `sudo snap install bluez  # version 5.48-3. `

Now, when I run `hciconfig`, nothing happens. I still get `rsp=$mgmtcode=$mgmterrestat=h11emsg='Invalid Index` after `le on` command.