IanHarvey/bluepy #222: Unable to build on HASS.IO raspberry pi

tomh05, 2017-09-23:
I've tried building this in homeassistant on a raspberry pi but get this error.
```
Unable to install package magicblue==0.4.3:
Command "/usr/bin/python3 -u -c "import setuptools, tokenize;file='/tmp/pip-build-mhr74ky0/bluepy/setup.py';f=getattr(tokenize, 'open', open)(file);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, file, 'exec'))" install --record /tmp/pip-98pxt0j5-record/install-record.txt --single-version-externally-managed --prefix --compile --user --prefix="
failed with error code 1 in /tmp/pip-build-mhr74ky0/bluepy/
```
Any ideas what's going wrong, or where I'd be able to get more detailed logs for debugging?

IanHarvey, 2017-09-25:
I can't really help with `magicblue`, so can you let me know what happens if you try to install bluepy on its own?

The README.md file has full instructions, but you'll need to `sudo apt-get install libbglib2.0-dev` first, and use `pip3 install` if installing for Python 3. 

Installation issues are currently being discussed at #158 .

Thanks
Ian

ArnaudLoos, 2017-09-28:
I too found my way here by trying to incorporate bluepy with homeassistant. I'm running HA with the All-In-One installer (on Debian) and I was able to add bluepy via pip no problem. The problem you may be facing is that Hass.io is based on ResinOS and Docker images and behaves differently than a standard OS. I've not played with Hass.io much yet because I do want a full OS underneath my HA installation for this very reason.

IanHarvey, 2017-10-22:
If the problem is installation related - I'd recommend trying to install `bluepy` from source as documented in the README.md file. There's ongoing discussion at #158 about installation problems. 

If you're running on a Raspberry Pi Zero, check out #223 - we currently have some issues with the helper not being correctly by `pip` on that platform.

If there are problems you suspect are Docker related, there was a thread at https://github.com/moby/moby/issues/16208 discussing similar issues.

Thanks
Ian

closed 2017-11-13 by IanHarvey