IanHarvey/bluepy #223: 'helper exited' + segfault

tedder, 2017-09-24:
Pi Zero (non-W) with a Redbear IOT PHat. 'lescan' works, but bluepy doesn't. I used the example in Scanner and ran it with sudo. Here's the tail of the stack:

```
  File "/usr/local/lib/python3.5/dist-packages/bluepy/btle.py", line 272, in _mgmtCmd
    rsp = self._waitResp('mgmt')
  File "/usr/local/lib/python3.5/dist-packages/bluepy/btle.py", line 304, in _waitResp
    raise BTLEException(BTLEException.INTERNAL_ERROR, "Helper exited")
bluepy.btle.BTLEException: Helper exited
```

I wondered if bluepy-helper was to blame, so I tried running it directly:
```
$ sudo /usr/local/lib/python3.5/dist-packages/bluepy/bluepy-helper
Segmentation fault
```

I tried installations on python3 and python2. How can I help debug this?

IanHarvey, 2017-09-24:
First question is where the `bluepy-helper` itself came from...
- I'm assuming this is the latest release (1.1.2). If not, can you try reinstalling this?
- Did you install it with `pip install` or by running a `setup.py` on the source? 
- Can you do a `ls -l /usr/local/lib/python3.5/dist-packages/bluepy/bluepy-helper` and `file /usr/local/lib/python3.5/dist-packages/bluepy/bluepy-helper` and post the results?

Thanks
Ian

tedder, 2017-09-24:
Thanks. I'm glad you replied so we can step through this.

* Latest release.
* Used `pip install`. I know it compiled the bluepy-helper because I had to install `libboost` and such as it failed the first time around.
* below.

```$ ls -l /usr/local/lib/python3.5/dist-packages/bluepy/bluepy-helper
-rwxr-xr-x 1 root staff 648888 Sep 24 03:04 /usr/local/lib/python3.5/dist-packages/bluepy/bluepy-helper
pi@redbear:~$ file /usr/local/lib/python3.5/dist-packages/bluepy/bluepy-helper
/usr/local/lib/python3.5/dist-packages/bluepy/bluepy-helper: ELF 32-bit LSB executable, ARM, EABI5 version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux-armhf.so.3, for GNU/Linux 3.2.0, BuildID[sha1]=9efb6c5c9af3ae8e284ffbc77bf3820a355d7d97, not stripped```

IanHarvey, 2017-09-25:
Something weird is going on if you needed `libboost` - there's no C++ in `bluepy-helper`. I don't suppose you still have (or can reproduce) the log of the build failing without it? On a Pi I've been building things with the standard gcc 4.9.2 which comes with Raspbian Jessie - what are you using?

Something else to try is finding out whether the pre-built `bluepy-helper` works on your system. If you can, go to https://pypi.org/project/bluepy/#files and manually download `bluepy-1.1.2-cp34-cp34m-linux_armv7l.whl`. You can then unzip it - it's literally just a zipfile - and you should find `bluepy-1.1.2.data/purelib/bluepy/bluepy-helper`. It's a standalone executable so there's nothing Python 3.5-specific about it - if it runs you could just move it into place and it should be fine.

Thanks
Ian

bjoernhoefer, 2017-09-30:
Hi,

I ran into the same error - and fixed it with the download of the bluepy-helper.

Additional Detail - I installed it on a raspberry pi 3 and a raspberry pi zero - and it only failed on the zero... maybe a coincidence

Thanks for the solution.

ukBaz, 2017-10-01:
It might not be a coincidence that it worked on the RPi3 and failed on the RPi Zero W.

As you can see from: https://en.wikipedia.org/wiki/Raspberry_Pi
Those two boards have different architectures on them (ARM8 vs ARM6) so will need different wheels

IanHarvey, 2017-10-02:
So I've managed to reproduce this on a Zero W, using latest Raspbian Stretch.

It's well weird - `pip install` appears to want to build it from source, but the thing it builds and installs is corrupted in some way I haven't yet fathomed. Building it from GitHub source, on the same box, with `setup.py build` seems to make a working `bluepy-helper`.

Maybe there's some way in which "installing" it from the build directory to `/usr/local/lib` is borking the file? More research needed...

ChristianKuehnel, 2017-10-17:
Hey guys, 

I have a similar problem but on a amd64 machine:
* I installed bluepy 1.1.2: ```sudo pip3 install bluepy```
* When I started bluepy-helper from the command line, I got this error saying the executable is broken:
> bash: ./bluepy-helper.corrupted: Kann die Binärdatei nicht ausführen: Fehler im Format der 
> Programmdatei
* After running ```sudo make``` in the installation directory manually, it worked fine.
* When looking at ```readelf bluepy-helper``` it tells me that I'm running the version for ARM v7, which obviously does not work for my amd64 machine:
> File Attributes
>  Tag_CPU_name: "7-A"
>  Tag_CPU_arch: v7

So I guess there is something wrong with the build/installation process and it tries to deploy the ARM v7 binary to all platforms...

I uploaded my 2 executables (broken and working):
[executables..tar.gz](https://github.com/IanHarvey/bluepy/files/1392482/executables.tar.gz)

jonsnow1357, 2017-10-20:
HI all,

I can confirm this bug, on a raspberry pi :(.
As suggested copying bluepy-handler from pypi.org seems to fix it.

I have uninstalled and reinstalled bluepy with '-v' and I have some additional information pasted below that I hope will be useful. Let me know if I can help with something else.

<pre>
[pi@rpi04 17:26:51]
[~]$ sudo pip -v install bluepy
Converted retries value: Retry(total=5, connect=None, read=None, redirect=None) -> Retry(total=Retry(total=5, connect=None, read=None, redirect=None), connect=None, read=None, redirect=None)
Converted retries value: Retry(total=5, connect=None, read=None, redirect=None) -> Retry(total=Retry(total=5, connect=None, read=None, redirect=None), connect=None, read=None, redirect=None)
Collecting bluepy
  1 location(s) to search for versions of bluepy:
  * https://pypi.python.org/simple/bluepy/
  Getting page https://pypi.python.org/simple/bluepy/
  Looking up "https://pypi.python.org/simple/bluepy/" in the cache
  No cache entry available
  Starting new HTTPS connection (1): pypi.python.org
  https://pypi.python.org:443 "GET /simple/bluepy/ HTTP/1.1" 200 816
  Updating cache with response from "https://pypi.python.org/simple/bluepy/"
  Caching b/c date exists and max-age > 0
  Analyzing links from page https://pypi.python.org/simple/bluepy/
    Found link https://pypi.python.org/packages/13/4f/b2f99f8c5b14ea1d9b2ff311fe0d232affc5a60a2c204825daa51703a358/bluepy-1.1.1.tar.gz#md5=037215b3c60e2297b3037c9f6cabafe8 (from https://pypi.python.org/simple/bluepy/), version: 1.1.1
    Found link https://pypi.python.org/packages/2f/5e/b0f10966bf23b98029164d034d3fc474f1f957885e660e48d61fa462a292/bluepy-1.0.5.tar.gz#md5=34c749540cf7b889c1c3d1b040afcdc6 (from https://pypi.python.org/simple/bluepy/), version: 1.0.5
    Skipping link https://pypi.python.org/packages/33/bd/b4ee0fb003917e21b377ec7af53bfad977fcf89f8700868368b000fde13e/bluepy-1.1.2-cp27-cp27mu-linux_armv7l.whl#md5=68b3847da7d2212b1267a0f6ee13bd56 (from https://pypi.python.org/simple/bluepy/); it is not compatible with this Python
    Skipping link https://pypi.python.org/packages/75/59/e26c0ce5d184e6172e323c1db882bc5b5e28ed20e34d08f2be90341f75b7/bluepy-1.1.2-cp34-cp34m-linux_armv7l.whl#md5=834c23d93af7554ff2c00bb4add29e52 (from https://pypi.python.org/simple/bluepy/); it is not compatible with this Python
    Found link https://pypi.python.org/packages/a2/23/38f57b2f84f99d9aa56efd20fbd0189ceb202f7a48028f634644b8524356/bluepy-1.0.3.tar.gz#md5=59abc628b1d851e18db8d11dfd20d9d2 (from https://pypi.python.org/simple/bluepy/), version: 1.0.3
    Found link https://pypi.python.org/packages/b7/6f/99900cbffd7714e75e433ad1ea594f6d0f9cdf9c8bbaffdfef3c5cbf8500/bluepy-1.0.4.tar.gz#md5=6c089943b2bd23c1e12262b28a872c71 (from https://pypi.python.org/simple/bluepy/), version: 1.0.4
    Found link https://pypi.python.org/packages/ea/45/911b293581760e04212fa1c3d4412ded8faa4b6c4e8e39acc4e1f18ae286/bluepy-1.0.2.tar.gz#md5=be53d5848d697ede54dfcca271313695 (from https://pypi.python.org/simple/bluepy/), version: 1.0.2
    Found link https://pypi.python.org/packages/f1/fa/4830ec2ab95baa0201151bf18d5c031d94c520bd05ffdf692573a5261fe3/bluepy-1.1.2.tar.gz#md5=65ef76a2b8538be6fb909e668e9354af (from https://pypi.python.org/simple/bluepy/), version: 1.1.2
    Found link https://pypi.python.org/packages/fe/1c/73f960e8a1e378e390139d0a81f6d8388f70b7eea79bb711910841911d0e/bluepy-1.1.0.tar.gz#md5=c72f0adde853149f6c60b6106ae3a4f5 (from https://pypi.python.org/simple/bluepy/), version: 1.1.0
  Using version 1.1.2 (newest of versions: 1.0.2, 1.0.3, 1.0.4, 1.0.5, 1.1.0, 1.1.1, 1.1.2)
  Looking up "https://pypi.python.org/packages/f1/fa/4830ec2ab95baa0201151bf18d5c031d94c520bd05ffdf692573a5261fe3/bluepy-1.1.2.tar.gz" in the cache
  No cache entry available
  https://pypi.python.org:443 "GET /packages/f1/fa/4830ec2ab95baa0201151bf18d5c031d94c520bd05ffdf692573a5261fe3/bluepy-1.1.2.tar.gz HTTP/1.1" 200 497925
  Downloading bluepy-1.1.2.tar.gz (497kB)
  Downloading from URL https://pypi.python.org/packages/f1/fa/4830ec2ab95baa0201151bf18d5c031d94c520bd05ffdf692573a5261fe3/bluepy-1.1.2.tar.gz#md5=65ef76a2b8538be6fb909e668e9354af (from https://pypi.python.org/simple/bluepy/)
    98% |############################### | 491kB 479kB/s eta 0:00:01  Updating cache with response from "https://pypi.python.org/packages/f1/fa/4830ec2ab95baa0201151bf18d5c031d94c520bd05ffdf692573a5261fe3/bluepy-1.1.2.tar.gz"
  Caching due to etag
    100% |################################| 501kB 81kB/s
  Running setup.py (path:/tmp/pip-build-7JtwM6/bluepy/setup.py) egg_info for package bluepy
    Running command python setup.py egg_info
    running egg_info
    creating pip-egg-info/bluepy.egg-info
    writing pip-egg-info/bluepy.egg-info/PKG-INFO
    writing top-level names to pip-egg-info/bluepy.egg-info/top_level.txt
    writing dependency_links to pip-egg-info/bluepy.egg-info/dependency_links.txt
    writing entry points to pip-egg-info/bluepy.egg-info/entry_points.txt
    writing manifest file 'pip-egg-info/bluepy.egg-info/SOURCES.txt'
    reading manifest file 'pip-egg-info/bluepy.egg-info/SOURCES.txt'
    writing manifest file 'pip-egg-info/bluepy.egg-info/SOURCES.txt'
  Source in /tmp/pip-build-7JtwM6/bluepy has version 1.1.2, which satisfies requirement bluepy from https://pypi.python.org/packages/f1/fa/4830ec2ab95baa0201151bf18d5c031d94c520bd05ffdf692573a5261fe3/bluepy-1.1.2.tar.gz#md5=65ef76a2b8538be6fb909e668e9354af
Building wheels for collected packages: bluepy
[... cut ...]
  Entry point address:               0x12049
  Start of program headers:          52 (bytes into file)
  Start of section headers:          647328 (bytes into file)
  Flags:                             0x5000400, Version5 EABI, hard-float ABI
  Size of this header:               52 (bytes)
  Size of program headers:           32 (bytes)
  Number of program headers:         9
  Size of section headers:           40 (bytes)
  Number of section headers:         39
  Section header string table index: 36
</pre>

IanHarvey, 2017-10-22:
I think the issue is ARM architecture-related. I've just tried the 'bad' executable built on my Pi Zero in a Pi 3, and it runs fine. The problem now is how to debug what's happening when `pip install` does its job - the only real variables are the versions of `gcc` and `pkg-config` which are running, or something weird in the environment.

IanHarvey, 2017-11-01:
Aha. It seems that the "source" tarball for 1.1.2 actually contains the `bluepy-helper` executable in it, and for reasons I fail to understand, `make -C ./bluepy clean` is not removing it (or it has already been installed before `make` gets there...)

IanHarvey, 2017-11-13:
I have made a new release (1.1.4) which (for me) fixes the problem.

More details are discussed in related issue #158.

Thanks
Ian

tedder, 2017-11-13:
replied on #158. Closing.

closed 2017-11-13 by tedder

jonsnow1357, 2017-11-14:
just updated to 1.1.4 on my raspberry pi and it seems to work.

Thank you :)