IanHarvey/bluepy #225: Detection of Advertising Data and Scan Response Data

Nppn7, 2017-10-09:
Hello,

I have a device being used as peripheral providing both advertising data and scan response data. Bluepy is being used as central. When I sniff the communication I can see that the peripheral is sending advertising and scan response data. When I try to see the same in the python scripts I get confusing behaviour. I adopted the sample code from [https://ianharvey.github.io/bluepy-doc/scanner.html](https://ianharvey.github.io/bluepy-doc/scanner.html) as follows:

```
from bluepy.btle import Scanner, DefaultDelegate

class ScanDelegate(DefaultDelegate):
    def __init__(self):
        DefaultDelegate.__init__(self)

    def handleDiscovery(self, dev, isNewDev, isNewData):
        if (dev.addr == "00:0b:57:15:9b:f2"):
            if isNewDev:
                print "Discovered device", dev.addr
                for (adtype, desc, value) in dev.getScanData():
                    print "  %s = %s" % (desc, value)
            elif isNewData:
                print "Received new data from", dev.addr
                for (adtype, desc, value) in dev.getScanData():
                    print "  %s = %s" % (desc, value)

scanner = Scanner().withDelegate(ScanDelegate())
devices = scanner.scan(10.0)
```
When I start scanning and the peripheral is discovered for the first time `isNewDev` is true. When data is updated `isNewData` is true. The peripheral is sending different info in advertising and scan response data. The problem is that on first discovery not advertising but scan response data is shown. And on updated values sometimes advertising data is shown and not scan response data. So most of the time the right statements are executed containing wrong data.

Does anyone know how this can be solved?

Jeffrey-Hall, 2018-03-08:
I am having this issue too. Is there any update?

Nppn7, 2018-03-12:
For me there is no update. I think the script isn't checking for whether the data is advertising or scan response. It is only checking whether data from a specific device is arriving for the first time or not. For checking advertising /scan response data I used [this](https://github.com/carsonmcdonald/bluez-experiments/blob/master/experiments/scantest.c). I modified the file by checking `info->length` to differ advertising and scan response.

pepelisu, 2019-03-20:
I also had the same problem. However it is solved using the option passive = True in the call to scan. 
```devices = scanner.scan(10.0, passive=True)```
The default mode is active and then getScanData returns the response data not the advertising data. If passive mode is used, getScanData returns the advertising data and not the response data.

Divya-nemuri, 2022-08-17:
> I also had the same problem. However it is solved using the option passive = True in the call to scan. `devices = scanner.scan(10.0, passive=True)` The default mode is active and then getScanData [...]

adding passive = True in  scanner.scan(10.0, passive=True) giving error