IanHarvey/bluepy #226: Bluez verison compatibility

rdvmikko, 2017-10-16:
Afaik bluepy and especially bluepy-helper are built against bluez 5.29. At the moment the latest bluez version seems to be 5.47. Are threre any plans / ongoing efforts to move to more recent bluez versions?

danielunderwood, 2018-02-23:
I was wondering about this too. I've created a fork and updated to 5.48 without any issues so far and can open up a PR after a bit more testing. But is there a reason for not linking to a locally installed bluepy?

IanHarvey, 2018-09-24:
The 1.2.0 release now uses BlueZ 5.47.  Bluez itself doesn't really provide a user-mode C library; there's just the kernel interface and a set of executables. They have a DBus API, but this was very immature at the time I started bluepy, and basically didn't work well enough.

If I ever get around to writing "bluepy 2.0" it will ditch the C code altogether, and will have a pure-Python implementation all the way down to the socket interface.

closed 2018-09-24 by IanHarvey

gandy92, 2018-09-24:
When planning the pure-python implementation, please keep in mind that execution of a python script with root privileges is undesirable. Currently, this can easily be avoided with a 
```
setcap 'cap_net_raw,cap_net_admin+eip' bluetooth-helper
```
(which needs to be repeated after each upgrade of the bluepy package - may I suggest this to be the installation default?)

IMHO, doing the same for the python interpreter as a whole may be almost as undesirable as running it with root privileges.