IanHarvey/bluepy #227: How to connect to peripheral device with pairing key?

Cooper-Weng, 2017-10-18:
I want to use RPI3 connect a peripheral device, but this device needs pairing key to establish connections.

When I want to connect the peripheral device use smartphone, it will bring out a dialogue box, and request pairing key.

But when  I use RPI3 want to connect the peripheral device, it just shows connection fail.

ukBaz, 2017-10-18:
How are you trying to pair the RPi3 to the peripheral?

I would suggest you test the pairing using `bluetoothctl` on the command line of the RPi3.
First I would suggest you remove the device. For example:
```
pi@RPi3:~ $ bluetoothctl 
[NEW] Controller B8:27:EB:22:57:E0 RPi3 [default]
[bluetooth]# devices 
Device D4:AE:95:4C:3E:A4 BBC micro:bit
[bluetooth]# remove D4:AE:95:4C:3E:A4 
[DEL] Device D4:AE:95:4C:3E:A4 BBC micro:bit
Device has been removed
[bluetooth]# exit
```
Once you have cleared the saved devices then you can do the pairing process. For example:
```
pi@RPi3:~ $ sudo service bluetooth restart
pi@RPi3:~ $ bluetoothctl 
[NEW] Controller B8:27:EB:22:57:E0 RPi3 [default]
[bluetooth]# agent KeyboardDisplay
Agent registered
[bluetooth]# agent on
Agent is already registered
[bluetooth]# scan on
Discovery started
[CHG] Controller B8:27:EB:22:57:E0 Discovering: yes
[NEW] Device D4:AE:95:4C:3E:A4 BBC micro:bit [zezet]
[bluetooth]# scan off
[BBC micro:bit [zezet]]# pair D4:AE:95:4C:3E:A4 
Attempting to pair with D4:AE:95:4C:3E:A4
[CHG] Device D4:AE:95:4C:3E:A4 Connected: yes
Request passkey
[CHG] Device D4:AE:95:4C:3E:A4 ServicesResolved: yes
[agent] Enter passkey (number in 0-999999): 133186
[CHG] Device D4:AE:95:4C:3E:A4 Paired: yes
Pairing successful
[CHG] Device D4:AE:95:4C:3E:A4 ServicesResolved: no
[CHG] Device D4:AE:95:4C:3E:A4 Connected: no
[bluetooth]# connect D4:AE:95:4C:3E:A4 
Attempting to connect to D4:AE:95:4C:3E:A4
[CHG] Device D4:AE:95:4C:3E:A4 Connected: yes
[CHG] Device D4:AE:95:4C:3E:A4 Name: BBC micro:bit
[CHG] Device D4:AE:95:4C:3E:A4 Alias: BBC micro:bit
Connection successful
[BBC micro:bit]# disconnect
[bluetooth]# exit
```

As pairing is a one time operation you should now be able to connect to the peripheral from the Python code.

abhishek-v-pandey, 2021-06-16:
Hello  @Cooper-Weng did you got the solution. Can you please explain it as I am facing the same issue.
@ukBaz I tried your method , its not working.
I have a BLE sensor and using Raspberry Pi 4.

Thanks!