IanHarvey/bluepy #228: setMTU() function not working?

Cooper-Weng, 2017-10-19:
My BLE peripheral device is "Nordic NRF52832".
And I refer to #113 reply, want to use the notification to receive more than 20 bytes at once, but it not working.

I have to change the **BLE_GATT_ATT_MTU_DEFAULT** value to **247** in "nRF5 SDK ble_gatt.h", and in RPI3 have to **setMTU(247)**, and RPI3 can receive more than 20 bytes.

But this setting is not a regular way and sometimes it caused "Nordic NRF52832" occur error event and disconnect.

mobilinkd, 2018-03-30:
Confirmed.  setMTU() works in the sense that the new MTU is properly negotiated with the device but fails because bluepy only sends and receives 20 bytes at a time after the MTU is changed.

After setting the MTU, I can see the MTU properly negotiated with my device and the device sending 100+ byte packets.  But the data provided to the Python application by bluepy is truncated at 20 bytes.

fellerts, 2018-05-06:
I have this issue as well, using the nRF52832. As mobilinkd said, I see that the MTU negotation happens as expected, and the device sends 100+ bytes, but all data received from bluepy is 20 bytes long.

Firefox2005, 2019-01-17:
Did anyone find a solution or a workaround for that? Sending my data in 20 bytes packages takes forever with a write with response.

wofcardoso, 2019-04-10:
I am facing the same problem, working with a Scale which responds with a 34 bytes, did anyone found a solution?