IanHarvey/bluepy #229: The write withResponse flag is backwards

tautology0, 2017-10-20:
The documentation (and variable name) stats that the characteristic.write withResponse flag is set to True if a response should be expected:

> Bluetooth LE allows the sender to
>     request the peripheral to send a response to confirm that the data has been received.
>     Setting the *withResponse* parameter to *True* will make this request.

But in the code, it does the opposite effect:

```
    if (with_response)
        gatt_write_char(attrib, handle, value, plen,
                    char_write_req_cb, NULL);
    else
    {
        gatt_write_cmd(attrib, handle, value, plen, NULL, NULL);
        resp_begin(rsp_WRITE);
        resp_end();
    }
```

Can we have an update to the documentation to reflect the real situation (and ideally a change to the variable name).

IanHarvey, 2017-10-22:
Good question. `gatt_write_char()` calls into `enc_write_req()` in BlueZ's _att.c_, which sends an `ATT_OP_WRITE_REQ` - this is the one with the response from the peripheral (Bluetooth 4.0 spec, Vol 3 part F, 3.4.5.1). 

`gatt_write_cmd()` calls `enc_write_cmd()` which uses `ATT_OP_WRITE_CMD`, the one without a peripheral response (3.4.5.3). 

So I think the code is right; the reason it looks like this is that `resp_begin()` etc. are functions to send a reply back from `bluepy-helper` to `btle.py`. For `gatt_write_char()`, the reply to `btle.py` is done in function `char_write_req_cb()`, which is called by the BlueZ stack when the peripheral sends a response. For `gatt_write_cmd()`, there is no peripheral response, so the code sends an immediate acknowledgement back to `btle.py`.

Are you finding problems using the _withResponse_ settings? If so, can you describe what you're doing and what went wrong?

Thanks
Ian

tautology0, 2017-10-23:
What drew me to raise this was trying to test a device which had a "WRITE NO RESPONSE" characteristic (I was using bleah rather than doing it raw; but I did delve down to the code level to work out why it was failing).

If the call was write(data, False) then it would not update the characteristic. If I mangled the code to do a write(data,True) it would successfully write the characteristic.

I was testing with a cheap ITAG, that had an "Alert Level" characteristic (UUID: 2a06) that raised and audible beep if the characteristic was written to.

I can give you some test scripts to demonstrate what I mean; but I can't do this until the end of the week as I don't have the device with me at the moment.

mdxs, 2020-10-21:
@tautology0 As you indicated yourself, this issue can be closed (as you indicated that it was hardware related).

closed 2020-10-21 by tautology0