IanHarvey/bluepy #230: ble write no response properties will blocking thread when you call writeCharacteristic

wuyinyao, 2017-10-21:
when you ble device no support response with write action, your current execute thread will be blocking,  root cause is in Peripheral object, the writeCharacteristic method will wait resp without timeout setting by default, self._getResp('wr'), after write, so the thread going block, next write request no response.
you can try to add timeout parameter  to fix this problem or don't call self._getResp('wr') methon when withResponse=False

NickWaterton, 2018-01-03:
I had the same issue, I added a timeout parameter as you mentioned.

Here is my diff on btle.py

```
302d301
<         if timeout is None: timeout = 60 #Added by Nick
488c487
<     def readCharacteristic(self, handle, timeout=20):
---
>     def readCharacteristic(self, handle):
490c489
<         resp = self._getResp('rd',timeout)
---
>         resp = self._getResp('rd')
498c497
<     def writeCharacteristic(self, handle, val, withResponse=False,timeout=20):
---
>     def writeCharacteristic(self, handle, val, withResponse=False):
503c502
<         return self._getResp('wr',timeout)
---
>         return self._getResp('wr')
```

edwios, 2019-01-21:
With Python3.5, adding time out didn't work and caused "RuntimeError concurrent poll() invocation".

The proper way, anyway, is for writeCharacteristic() to return None when withResponse is False, else return _getResp('wr').

If the characteristic does not support write response or you don't need it, then just call writeCharacteristic(self, handle, val), otherwise, call writeCharacteristic(self, handle, val, True). Neither cases shall block under normal circumstance.

On the other hand, timeout shall function properly for dealing with abnormal cases such as the "expected' wr response got lost in the air. For this, I don't think using readline() (btle.py line 347) is wise as it will never return until it got an EOL which may never happen.

Hedroed, 2021-12-12:
up