IanHarvey/bluepy #231: SensorTag update interval

karray, 2017-10-22:
I'm using Raspberry PI 3 to collect data from a SensorTag. No matter what I set as `tag.waitForNotifications()`, sensor's data updates every second.

Can you please point me to the right function or documentation.

johnp789, 2018-04-03:
Generally, the SensorTag firmware accepts measurement periods for each sensor, with 1s being the default period.  Writing a different value to the period characteristic can change the period.  For example, the HDC1000 period can be set between 100ms and 2.55s, per the [TI User's Guide entry](http://processors.wiki.ti.com/index.php/CC2650_SensorTag_User%27s_Guide#Humidity_Sensor).

karray, 2018-04-13:
I've achieved it by writing a characteristic like this:

```
# get MovementSensor period characteristic
gp = tag.getCharacteristics(uuid='f000aa83-0451-4000-b000-000000000000')
# set 100ms (20/10 = 0x2 in hex)
gp[0].write(struct.pack("B", 0xa))
```

closed 2018-04-13 by karray