IanHarvey/bluepy #233: How to exit gracefully

alcir, 2017-11-11:
Sorry I'm not a Python expert.
But how to exit gracefully when pressing ctrl-c or killing the program?
I mean that I want to perform a device.disconnect() or write a value in a descriptor when I stop the program.
At the moment, I get

<pre>
File "8gat.py", line 59, in <module>
    p.disconnect()
  File "/home/alessio/.local/lib/python2.7/site-packages/bluepy/btle.py", line 417, in disconnect
    self._getResp('stat')
  File "/home/alessio/.local/lib/python2.7/site-packages/bluepy/btle.py", line 369, in _getResp
    resp = self._waitResp(wantType + ['ntfy', 'ind'], timeout)
  File "/home/alessio/.local/lib/python2.7/site-packages/bluepy/btle.py", line 304, in _waitResp
    raise BTLEException(BTLEException.INTERNAL_ERROR, "Helper exited")
bluepy.btle.BTLEException: Helper exited
</pre>

trocotronic, 2018-01-09:
I have the same problem. Is there any fix?

intrepiddigital, 2018-01-09:
I am also not an expert but I have 

try......
.... code
except Keyboardinterrupt
     print("Cnt + C detected")    
     sys.exit()

Don't know if it could be called graceful but at least you don't get a traceback

intrepiddigital, 2018-01-09:
Don't forget to indent

try......
.... code
except Keyboardinterrupt
    print("Cnt + C detected")
    sys.exit()

mobilinkd, 2018-03-30:
Already opened at https://github.com/IanHarvey/bluepy/issues/191

mdxs, 2020-10-21:
@alcir As #264 has been merged (see that PR for how to handle the KeyboardInterrupt); this issue can be closed.