IanHarvey/bluepy #248: response: {'code': ['protoerr'], 'rsp': ['mgmt']}

brianjmurrell, 2017-12-27:
I'm trying a little test program just to test basic functionality:

```
from bluepy.btle import Scanner, DefaultDelegate

class ScanDelegate(DefaultDelegate):
    def __init__(self):
        DefaultDelegate.__init__(self)

    def handleDiscovery(self, dev, isNewDev, isNewData):
        if isNewDev:
            print "Discovered device", dev.addr
        elif isNewData:
            print "Received new data from", dev.addr

scanner = Scanner().withDelegate(ScanDelegate())
devices = scanner.scan(10.0)

for dev in devices:
    print "Device %s (%s), RSSI=%d dB" % (dev.addr, dev.addrType, dev.rssi)
    for (adtype, desc, value) in dev.getScanData():
        print "  %s = %s" % (desc, value)
```

But when I run it I get:

```
$ python blpy-test.py
writing: le on
response: {'code': ['protoerr'], 'rsp': ['mgmt']}
Traceback (most recent call last):
  File "blpy-test.py", line 14, in <module>
    devices = scanner.scan(10.0)
  File "/home/brian/igrill/venv/lib/python2.7/site-packages/bluepy/btle.py", line 681, in scan
    self.start(passive=passive)
  File "/home/brian/igrill/venv/lib/python2.7/site-packages/bluepy/btle.py", line 619, in start
    self._mgmtCmd("le on")
  File "/home/brian/igrill/venv/lib/python2.7/site-packages/bluepy/btle.py", line 278, in _mgmtCmd
    "Failed to execute mgmt cmd '%s'" % (cmd))
bluepy.btle.BTLEException: Failed to execute mgmt cmd 'le on'
```

Notice the two lines of debug I added to _mgmtCmd() above:

```
writing: le on
response: {'code': ['protoerr'], 'rsp': ['mgmt']}
```

Any solutions to this error?

intrepiddigital, 2018-01-09:
Just a thought but the trace you have is identical to not running your code as root. "sudo python....

intrepiddigital, 2018-01-09:
but then I didn't read your words below the trace. !!!

sarvesh-80, 2018-03-13:
run using sudo
sudo python blpy-test.py

IanHarvey, 2018-07-23:
It's a property of the Linux Bluetooth kernel interface that only root (or a process with the right capabilities on some distributions) can do global operations on an HCI device, like starting scanning. So you need to run `blescan` and similar things as root.

closed 2018-07-23 by IanHarvey