IanHarvey/bluepy #253: BLE disconnected when call writeCharacteristic in notify example

grahaminnovations, 2018-01-22:
Hi, 
The code I used is based on the ble notify example. 
I have a separate function to deal with the GPIO button interrupt response. 
It will call the writeCharacteristic and send a short command to the peripheral device. 
The peripheral device will send back an ACK notification. 
Sometimes it cause the ble to disconnect. Any idea why is that?
```
class BleNotify():
    def __init__(self):
        self.hci_port = conf["ble_hci_port"]
        self.service_uuid = conf["ble_service_uuid"]
    
    def getConn(self):
        try:
            return self.conn
        except:
            return
        
    def connect_BLE(self, MACaddr, MACtype):
        try:
            self.conn = btle.Peripheral(MACaddr, MACtype)
            try:
                self.conn.setDelegate(NotifyDelegate())
            except:
                print('fail to set the delegate')
        except:
            print('fail to connect to: %s' % MACaddr)
            try:
                self.close_BLE()
                self.conn = False
            except:
                print('BLE has already disconnected')

    def notify_Init(self):
        try:
            # Find the service
            self.svc = self.conn.getServiceByUUID(self.service_uuid)
            # Find the notification characteristics: [0] write, [1] notify
            ch_Notify = self.svc.getCharacteristics()[1]
            # Turn on the notification alert
            self.conn.writeCharacteristic(ch_Notify.valHandle+1, b'\x01\x00', withResponse=True)
        except:
            print('Service request fails')

    def notify_Write(self, text):
        # Find the notification characteristics: [0] write, [1] notify
        ch_Write = self.svc.getCharacteristics()[0]
        self.conn.writeCharacteristic(ch_Write.valHandle, text)
        print("Sent: ", text)

    def notify_Wait(self):
        return self.conn.waitForNotifications(1.0)

    def close_BLE(self):
        self.conn.disconnect()

class NotifyDelegate(btle.DefaultDelegate):
    def __init__(self):
        btle.DefaultDelegate.__init__(self)
        
    def handleNotification(self, cHandle, data):
        DBG("Data Recv:", data)
    
def main():
    try:
        b = BleNotify()
        while True:
            if b.getConn():
                try:
                    b.notify_Init()
                    while True:
                        # Wait for any notification
                        if b.notify_Wait():
                            continue
                except:
                    DBG("BLE Disconnected")
                    try:
                        b.close_BLE()
                    except:
                        pass
                    b.connect_BLE(conf["ble_mac_addr"], conf["ble_mac_type"])
                    
            else:
                b.scan_BLE(conf["ble_scan_timeout"])
                time.sleep(1)

    except KeyboardInterrupt:
        try:
            b.close_BLE()
        except:
            pass

if __name__ == '__main__':
    main()
```

grahaminnovations, 2018-01-23:
The error message:
```
Traceback (most recent call last):
  File "/home/pi/GI-bleNotify/demo.py", line 111, in <module>
    if p.waitForNotifications(1.0):
  File "/usr/local/lib/python3.4/dist-packages/bluepy/btle.py", line 516, in waitForNotifications
    resp = self._getResp(['ntfy','ind'], timeout)
  File "/usr/local/lib/python3.4/dist-packages/bluepy/btle.py", line 369, in _getResp
    resp = self._waitResp(wantType + ['ntfy', 'ind'], timeout)
  File "/usr/local/lib/python3.4/dist-packages/bluepy/btle.py", line 337, in _waitResp
    raise BTLEException(BTLEException.INTERNAL_ERROR, "Unexpected response (%s)" % respType)
bluepy.btle.BTLEException: Unexpected response (wr)```

Thomas-Dimos, 2018-03-23:
I received the same error and as it looks when we are waiting for notifications and in my case another thread is going to write a value to a characteristic, a response is produced, as the error says ,and in the _waitResp() in the source code of btle.py which is called from waitForNotifications(),the last else is fired and produces this error.The thing is that in my case the write operation is always executed but then this error comes up.I don't know if it is too simple but maybe in the last else of this function if the response type is 'wr' then you can just skip it and wait again for the correct response instead of throwing an exception,but since we can't modify the code we hope it will be fixed

ronnymajani, 2019-02-26:
If you're using multithreading or some setup where multiple BLE functions can be called in "parallel"
then what can happen is that the write() function will send a write command, and then wait to receive the write response "wr".
One possible problem is, if there's another thread that's also waiting for a response (like calling waitForNotifications()) then when that response 'wr' is received, it can mistakenly get passed to waitForNotifications instead, and that function isn't expecting such a response type and will throw an error.
One way to workaround this is to call write() with the withResponse argument set to True (.write(value, withResponse=True) ), which will queue the write request instead so you get everything in the right order? (not exactly sure what's going on in this case, I didn't bother to go deeper than this).

ronnymajani, 2019-02-26:
Oh wait, no no ignore that I found the problem:
int the btle.py
the following code snippet has an indentation error
replace:
```python
def _getResp(self, wantType, timeout=None):
        if isinstance(wantType, list) is not True:
            wantType = [wantType]

        while True:
            resp = self._waitResp(wantType + ['ntfy', 'ind'], timeout)
            if resp is None:
                return None

            respType = resp['rsp'][0]
            if respType == 'ntfy' or respType == 'ind':
                hnd = resp['hnd'][0]
                data = resp['d'][0]
                if self.delegate is not None:
                    self.delegate.handleNotification(hnd, data)
            if respType not in wantType:
                continue
            return resp
```
with:
```python
def _getResp(self, wantType, timeout=None):
        if isinstance(wantType, list) is not True:
            wantType = [wantType]

        while True:
            resp = self._waitResp(wantType + ['ntfy', 'ind'], timeout)
            if resp is None:
                return None

            respType = resp['rsp'][0]
            if respType == 'ntfy' or respType == 'ind':
                hnd = resp['hnd'][0]
                data = resp['d'][0]
                if self.delegate is not None:
                    self.delegate.handleNotification(hnd, data)
            if respType not in wantType:
                continue
            return resp
```

the part that's suppose to continue the loop if the expected response isn't received yet, is mistakenly indented one more tab, making it a redundant statement that never gets executed.
Just make that fix and all will be good.
I'm gonna submit a pull request soon.

ronnymajani, 2019-02-26:
Ok, well it seems that my first comment was more accurate,
the issue still presists, and it definitely seems to be a multithreading issue where the 'wr' response is propagated to waitForNotifications() instead of .write().
Like I said, just set withResponse to True and you should be fine. The only downside is a slowdown due to using write with response.