IanHarvey/bluepy #254: (tag, tval) = item.split('=') ValueError: need more than 1 value to unpack

manuel-masiello, 2018-01-30:
Hello,

I have an error with Bluepy:

```

Sent:  svcs

Got: "01800-0000-1000-8000-00805f9b34fb hstart=h28 hend=hFFFF uuid='0000180f-0000-1000-8000-00805f9b34fb\n"
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "main.py", line 28, in run
    if connection.waitForNotifications(1):
  File "/usr/local/lib/python2.7/dist-packages/bluepy/btle.py", line 516, in waitForNotifications
    resp = self._getResp(['ntfy','ind'], timeout)
  File "/usr/local/lib/python2.7/dist-packages/bluepy/btle.py", line 369, in _getResp
    resp = self._waitResp(wantType + ['ntfy', 'ind'], timeout)
  File "/usr/local/lib/python2.7/dist-packages/bluepy/btle.py", line 317, in _waitResp
    resp = BluepyHelper.parseResp(rv)
  File "/usr/local/lib/python2.7/dist-packages/bluepy/btle.py", line 282, in parseResp
    (tag, tval) = item.split('=')
ValueError: need more than 1 value to unpack

Got: "rsp=$find hstart=h1 hend=h5 uuid='00001801-0000-1000-8000-00805f9b34fb hstart=h14 hend=h1C uuid='000rsp=$ind hnd=h3 d=b0100FFFF\n"
Traceback (most recent call last):
  File "main.py", line 58, in <module>
    print p.getServices()
  File "/usr/local/lib/python2.7/dist-packages/bluepy/btle.py", line 440, in getServices
    return self.services
  File "/usr/local/lib/python2.7/dist-packages/bluepy/btle.py", line 436, in services
    self._serviceMap = self.discoverServices()
  File "/usr/local/lib/python2.7/dist-packages/bluepy/btle.py", line 422, in discoverServices
    rsp = self._getResp('find')
  File "/usr/local/lib/python2.7/dist-packages/bluepy/btle.py", line 369, in _getResp
    resp = self._waitResp(wantType + ['ntfy', 'ind'], timeout)
  File "/usr/local/lib/python2.7/dist-packages/bluepy/btle.py", line 317, in _waitResp
    resp = BluepyHelper.parseResp(rv)
  File "/usr/local/lib/python2.7/dist-packages/bluepy/btle.py", line 282, in parseResp
    (tag, tval) = item.split('=')
ValueError: too many values to unpack
Sent:  disc

Got: "rsp=$stat state=$disc mtu=h0 sec='low\n"
Stopping  /usr/local/lib/python2.7/dist-packages/bluepy/bluepy-helper

```

I see 3 errors: 

1 - "need more than 1 value to unpack"
It's because self._helper.stdout.readline() return this string:

`01800-0000-1000-8000-00805f9b34fb hstart=h28 hend=hFFFF uuid='0000180f-0000-1000-8000-00805f9b34fb\n`

The key for 01800******* is not set and (tag, tval) = item.split('=') don't have 2 elements.

2 - "too many values to unpack"

self._helper.stdout.readline() return this string:

`rsp=$find hstart=h1 hend=h5 uuid='00001801-0000-1000-8000-00805f9b34fb hstart=h14 hend=h1C uuid='000rsp=$ind hnd=h3 d=b0100FFFF\n`

There are a mistake here: `uuid='000rsp=$ind` an blank space is missing.

3 - Simple cote error:

`uuid='0000180f-0000-1000-8000-00805f9b34fb\n`

The value of uuid start by a cote but never close...

This sounds like reading problems on the port. All these errors are systematic.

Here is how I have the error:

I use Python on my RPI3 and I search to connect Virtual Peripherals on "LightBlue explorer" iOS Apps with the "time" virtual peripherals.

And my code:

```
from bluepy.btle import Scanner, DefaultDelegate, Peripheral
import threading

class NotificationDelegate(DefaultDelegate):
	def __init__(self, number):
		DefaultDelegate.__init__(self)
		self.number = number

	def handleNotification(self, cHandle, data):
		print 'Notification: Connection:' + str(self.number) + ' --- Handler:' + str(cHandle) + ' --- Msg:' + str(data)

connections = []
connection_threads = []
scanner = Scanner(0)

class ConnectionHandlerThread(threading.Thread):
	def __init__(self, connection_index):
		threading.Thread.__init__(self)
		self.connection_index = connection_index

	def run(self):
		connection = connections[self.connection_index]
		connection.setDelegate(NotificationDelegate(self.connection_index))
		while True:
			if connection.waitForNotifications(1):
				connection.writeCharacteristic(37, 'Thank you for the notification!')

bConnected = False
while not bConnected:
	print'Connected: ' + str(len(connection_threads))
	print 'Scanning...'
	devices = scanner.scan(2)
	for d in devices:

		bFound = False
		for aScanData in d.getScanData():
			#print aScanData
			if 'Complete Local Name' in aScanData:
				print aScanData[2]

				if aScanData[2] == 'Time':
					print('Found '+aScanData[2]+' at '+d.addr)
					bFound = True

		if bFound:
			p = Peripheral(d)
			connections.append(p)
			t = ConnectionHandlerThread(len(connections) - 1)
			t.start()
			connection_threads.append(t)
			bConnected = True

			print('Connected !!!')
			print p.getServices()

while True:
	pass
```

Thx
Numael

manuel-masiello, 2018-01-30:
I have the same behavior on my PC Ubuntu with BLE SIM Android (Galasy S8).

```
Connected: 0
Scanning...
Galaxy S8
Found Galaxy S8 at 70:96:28:90:21:34
Connected !!!
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "main.py", line 28, in run
    if connection.waitForNotifications(1):
  File "/home/me/.local/lib/python2.7/site-packages/bluepy/btle.py", line 516, in waitForNotifications
    resp = self._getResp(['ntfy','ind'], timeout)
  File "/home/me/.local/lib/python2.7/site-packages/bluepy/btle.py", line 369, in _getResp
    resp = self._waitResp(wantType + ['ntfy', 'ind'], timeout)
  File "/home/me/.local/lib/python2.7/site-packages/bluepy/btle.py", line 317, in _waitResp
    resp = BluepyHelper.parseResp(rv)
  File "/home/me/.local/lib/python2.7/site-packages/bluepy/btle.py", line 282, in parseResp
    (tag, tval) = item.split('=')
ValueError: need more than 1 value to unpack

Traceback (most recent call last):
  File "main.py", line 58, in <module>
    print p.getServices()
  File "/home/me/.local/lib/python2.7/site-packages/bluepy/btle.py", line 440, in getServices
    return self.services
  File "/home/me/.local/lib/python2.7/site-packages/bluepy/btle.py", line 436, in services
    self._serviceMap = self.discoverServices()
  File "/home/me/.local/lib/python2.7/site-packages/bluepy/btle.py", line 422, in discoverServices
    rsp = self._getResp('find')
  File "/home/me/.local/lib/python2.7/site-packages/bluepy/btle.py", line 369, in _getResp
    resp = self._waitResp(wantType + ['ntfy', 'ind'], timeout)
  File "/home/me/.local/lib/python2.7/site-packages/bluepy/btle.py", line 317, in _waitResp
    resp = BluepyHelper.parseResp(rv)
  File "/home/me/.local/lib/python2.7/site-packages/bluepy/btle.py", line 282, in parseResp
    (tag, tval) = item.split('=')
ValueError: too many values to unpack
```

mkeesey, 2019-03-18:
I've noticed the `too many values to unpack` error when talking to avi-on devices as well. The problem seems to go away when falling back from bluepy `v/1.3.0` to `v/1.2.0`.

Code that changed around the area in 1.3.0 seems to be in b8ddd8f96e846b213ce57c1a5abe6251d8639861, but I don't know nearly enough about the bluetooth protocol to know what the changes should be doing.