IanHarvey/bluepy #259: Getting invalid start byte which Kills process

3-Tcubed, 2018-03-17:
I have about 30 BT devices being seen, the error dump looks like this, the bad byte can be any non-ascii value and any where in the data

Python 3.4.2 (default, Oct  8 2014, 14:38:51) 
[GCC 4.9.1] on linux

Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.4/threading.py", line 920, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.4/threading.py", line 868, in run
    self._target(*self._args, **self._kwargs)
  File "/home/pi/Downloads/welloeye/welloeye.py", line 643, in getScanData
    scanner.process(20)
  File "/usr/local/lib/python3.4/dist-packages/bluepy/btle.py", line 669, in process
    self.delegate.handleDiscovery(dev, (dev.updateCount <= 1), isNewData)
  File "/home/pi/Downloads/welloeye/welloeye.py", line 486, in handleDiscovery
    for (adtype, desc, value) in dev.getScanData():
  File "/usr/local/lib/python3.4/dist-packages/bluepy/btle.py", line 601, in getScanData
    for sdid in self.scanData.keys() ]
  File "/usr/local/lib/python3.4/dist-packages/bluepy/btle.py", line 601, in <listcomp>
    for sdid in self.scanData.keys() ]
  File "/usr/local/lib/python3.4/dist-packages/bluepy/btle.py", line 594, in getValueText
    return val.decode('utf-8')
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xa9 in position 4: invalid start byte

Might I suggest changing getValueText to trap error:

    def getValueText(self, sdid):
        val = self.scanData.get(sdid, None)
        if val is None:
            return None
        if (sdid==8) or (sdid==9):
            try:
                return val.decode('utf-8')
            except OSError as e:
                return binascii.b2a_hex(val).decode('utf-8')
        else:
            return binascii.b2a_hex(val).decode('utf-8')

Thorsten-Sick, 2018-06-13:
I have the same problem. Fixed it and tested it. I will create a PR now.
The problem is caused by "creative" device names. I also have several of those within scanning range.

closed 2018-06-16 by IanHarvey

souellet-ms, 2018-08-02:
Sorry for replying to this closed issue, but the fixes associated with it are probably preventing the reported error, but have caused me headache.

Here's my situation. My application is scanning for device indefinitely. For every new device found, I check that the short name or long name match a specific pattern. First, I checked that the short name and long name are not None. This is where this fix is invalid.

`            
shortName = scanEntry.getValueText(0x08)
longName = scanEntry.getValueText(0x09)

if longName is not None:
    self._delegate.handle_device(scanEntry.addr, longName)
elif shortName is not None:
    self._delegate.handle_dvice(scanEntry.addr, shortName)
`
Using 1.1.4, the name is either a type('unicode') or a type('NoneType') depending if the name is set or not. 

Now, using 1.2.0, the name is always a type('str') even if a name is not set, which returns 'None' as a string instead of a NoneType.

For now, I will have to use bluepy 1.1.4 until I found time to fix this issue or someone else does.

commit a3759e1d (2018-06-16): Handling for broken device names. Closes #259
commit e4911750 (2018-06-16): Merge pull request #275 from Thorsten-Sick/device_name_encoding_fix

Handling for broken device names. Closes #259