IanHarvey/bluepy #262: Passive scan returns different address type than active scan for BLE devices

celosiag, 2018-03-26:
Hi,

First, thank you for your work. I really enjoy this Bluetooth python module and its simplicity of use.

Unfortunately, I've discovered that, with passive scan, a found BLE device doesn't have the same address type than if it is discovered during an active scan. I checked the scanned results with Wireshark and it seems that the problem comes from the bluepy python module as Wireshark detects the same device address type whether an active or a passive scan is triggered.

Actually, I'm using the original python code from scanner.py (available in bluepy/bluepy/scanner.py) and I just changed the lines:

- print(strftime("%Y-%m-%d %H:%M:%S", gmtime()), dev.addr, dev.getScanData())
BY  print(strftime("%Y-%m-%d %H:%M:%S", gmtime()), dev.addr, dev.addrType, dev.getScanData())

- scanner.scan(10.0, passive=True) BY scanner.scan(10.0)

To illustrate my issue, for instance, when I run an active scan (with scanner.scan(10.0)), I have the following : ... 11:22:33:44:55:66 random.... And, you would have understood it, when I run a passive scan (with scanner.scan(10.0,passive=True)), I have the following : ... 11:22:33:44:55:66 public...

So please, can you fix this little problem ? Or maybe I can help you to do this if you can show me the code part involved in this process of finding device address type ?

Thanks in advance.

G.C.

gandy92, 2018-05-19:
Hi, 

you want to take a look at hci_monitor_cb() in bluepy-helper.c. I can try to look into this in the next few weeks, unless you'd like to debug it yourself. 

Best, Andy.

expandable, 2018-05-20:
Hi,
I committed a possible fix to https://github.com/gandy92/bluepy. Unfortunately I don't have any BLE devices advertising with random address, so I cannot test it. May I ask you to to verify the solution before I create a pull request?
Cheers,
Andy.

gandy92, 2018-05-20:
Hi,
I committed a possible fix to https://github.com/gandy92/bluepy. Unfortunately I don't have any BLE devices advertising with random address, so I cannot test it. May I ask you to to verify the solution before I create a pull request?
Cheers,
Andy.

celosiag, 2018-05-20:
Andy,

I've just tried your fix with a BLE device which advertises with random address. I've the pleasure to announce you that it works well ! Both active and passive scans show that advertising address used is random for this device. Thank you really much for your good job :).

Cheers,
G.C.

gandy92, 2018-05-20:
Great, thanks for the feedback. 
Cheers, 
Andy.

commit 15620fed (2018-06-16): Fix #262: Return correct address type in passive scan mode
commit 91227b73 (2018-06-16): Merge pull request #271 from gandy92/master

Fix #262: Return correct address type in passive scan mode
closed 2018-06-16 by IanHarvey