IanHarvey/bluepy #263: Vendor Service UUID (128b) is Reversed in Scan Results

mobilinkd, 2018-03-30:
The output of my scan shows the following:

    (6, 'Incomplete 128b Services', u'33c3695141abaab2484cd69001004a42')

That should match this service UUID

    SERVICE_UUID = UUID('424a0001-90d6-4c48-b2aa-ab415169c333')

but does not because the UUID is reversed. The service UUID shows up correctly in the  ``nRF Connect`` tool on Android, and this is the output I see from bluetoothctl and what I expect from bluepy on Linux:

    [bluetooth]# info EC:8D:39:A3:48:96
    Device EC:8D:39:A3:48:96
        ...
	    UUID: Vendor specific           (424a0001-90d6-4c48-b2aa-ab415169c333)

All of the related characteristic UUIDs are shown properly.  And I need to use the proper SERVICE_UUID (shown above) when calling ``getServiceByUUID()``.

Am I misunderstanding the scan results I am receiving in some way?

kesslera, 2018-04-19:
I've got the same issue. I want to filter the discovered devices by their advertised service uuid. After checking the `scanEntry.getScanData()`, the service uuid is in reversed order. 
Is is a bug in the library?

IanHarvey, 2018-07-23:
Should be fixed in release 1.2.0.

closed 2018-07-23 by IanHarvey