IanHarvey/bluepy #272: RPi3 connects to device via bluetoothctl but not via Bluepy

quique123, 2018-05-22:
I have this issue where Im trying to connect a RPi3 (raspbian and bluepy) to a BLE relay board (wired-RB).  The same RPi3 connects perfectly via bluepy script and sends commands to the service characteristic just fine on another test unit relay board (test-unit) IDENTICAL to the wired-unit that Im trying to communicate with now.

The py script is:

```
import bluepy.btle as btle
p = btle.Peripheral("00:0E:0B:00:75:12", "random")
s = p.getServiceByUUID("0000ffe0-0000-1000-8000-00805f9b34fb")
c = s.getCharacteristics()[0]
c.write("o", "utf-8")
p.disconnect()
```

But when I run the script I get this error:

> File "/usr/local/lib/python2.7/dist-packages/bluepy/btle.py", line 449, in getServiceByUUID
> raise BTLEException(BTLEException.GATT_ERROR, "Service %s not found" % (uuid.getCommonName()))
> bluepy.btle.BTLEException: Service ffe0 not found

I've been digging around but cant find an answer or reason.  I can see that exact same service on that exact same mac device from the bluetoothctl, but for some reason it just wont find it when using the bluepy python code.

quique123, 2018-05-29:
After hours of inspecting the btle.py file, I noticed the getServices() function was not being called.  I added a call to it and now it find can find the service:

    #!/usr/bin/env python
    import bluepy.btle as btle
    p = btle.Peripheral("00:0E:0B:00:75:12", "random")
    services=p.getServices()
    for service in services:
       print service
    s = p.getServiceByUUID("0000ffe0-0000-1000-8000-00805f9b34fb")
    c = s.getCharacteristics()[0]
    c.write("e", "utf-8")
    p.disconnect()

Now Im left with the issue that the BLE device is not responding to the command for some reason, but at least I dont get any py errors.

henry-stann, 2018-06-05:
I get this same error. I have an HM-10 connected to an arduino sending out "I am groot" every second, which I can read from an iPhone connected via bluetooth. 
What's more, is if I read, I am presented with 
| <newline and/or carriage return> _@x

If I use the commands:
bluetoothctl
connect 7C:01:0A:5F:40:58
menu gatt
select-attribute 0000ffe1-0000-1000-8000-00805f9b34fb
read

Then I am presented with:
|.._@x

I have not figured out how to write thru bluetoothctl. I hope my extension of this question may help some...

henry-stann, 2018-06-09:
If anyone's monitoring this forum, how did you install Bluez? This may be the issue (I'm redoing my system to test myself)

I used 
```
$cd bluez-5.49
$./configure --prefix=/usr --mandir=/usr/share/man --sysconfdir=/etc --localstatedir=/var
$make
$sudo make install

```

henry-stann, 2018-06-22:
Thanks Andy! I have yet to try this out; sorry.