IanHarvey/bluepy #274: Problem with writing characteristic value larger than 20 bytes

Piotr-Romaniuk, 2018-06-07:
Hi,
I have a problem with writing more than 20 bytes to characteristic via bluepy.
It works fine when I am doing this in bluetoothctl.
Is there some limitation?
Below is the code that I use.
Am I doing something wrong?
Regards,
Piotr

`  1 import sys
  2 from bluepy.btle import UUID, Peripheral
  3 
  4 
  5 
  6 
  7
  8
  9 srv_uuid = UUID("48b49446-4aab-11e8-842f-0fd5f89f718b")
 10
 11 fragment_uuid    = UUID("48b4987e-4aab-11e8-842f-0fd5f89f718b")
 12
 13 if len(sys.argv) != 2:
 14   print "ERROR, incorrect argument number."
 15   print "  usage", sys.argv[0], "bt-dev-addr"
 16   quit()
 17
 18 p = Peripheral(sys.argv[1], "random")
 19 upgrade_service = p.getServiceByUUID( srv_uuid )
 20
 21
 22 try:
 23     ch_fragm = upgrade_service.getCharacteristics( fragment_uuid )[0]
 24
 25     data20 = '\x01\x02\x03\x04\x05\x06\x07\x08\x09\x10\x11\x12\x13\x14\x15\x16\x17\x18\x19\x20'
 26     data30 = data20 + '\x21\x22\x23\x24\x25\x26\x27\x28\x29\x30'
 27
 28     print len( data20 )
 29     print len( data30 )
 30 #    ch_fragm.write( data20, True ) #This works
 31     ch_fragm.write( data30, True )
 32
 33
 34 finally:
 35     p.disconnect()
 36
`

froqstar, 2018-06-22:
Bluetooth LE 4.0 doesn't support more than 20 bytes per message, so this is a protocol limitation.
See this answer: 
https://stackoverflow.com/a/38914831

Piotr-Romaniuk, 2018-06-22:
Hi,
Thank you for pointing me the post with limitation discussion.

First tests of my device I performed with bluetoothctl (interactive linux program), it read more that 20 bytes, hence I expected the same behavior from bluepy.

The device had BlueNRG-2 chip that is Bluetooth 5.0. 
So the reason of the issue is that bluepy supports only 4.0, right?

dnsglk, 2021-05-19:
I have two nrf52840 chips, and both of them support BLE 5.0.  I also can send and receive more than 20 bytes using `bluetoothctl`. But can't do it with `bluepy`.