IanHarvey/bluepy #276: Should have more example and usecase?

avatarr, 2018-06-17:
Hi, i'm a new with bluepy and BLE by python. I want to share my experience in 2 days of work with bluepy and request about update Docs

I found some method not written in the Docs and Docs are unclear. I tried to retrieve data from service and characteristics like 
[image]

with code 
```python 
rom bluepy.btle import DefaultDelegate, Peripheral

class NotificationDelegate(DefaultDelegate):
    def __init__(self):
        DefaultDelegate.__init__(self)

    def handleNotification(self, cHandle, data):
        print("A notification was received: %s" %data)

p = Peripheral("00:1C:05:FF:CA:8D", "public")

p.setDelegate( NotificationDelegate() )

svc = p.getServiceByUUID("1822")
ch = svc.getCharacteristics("2a5f")[0]
p.writeCharacteristic(ch.valHandle + 1, "\x01\x00", withResponse=True)

while True:
    if p.waitForNotifications(1.0):
        # handleNotification() was called
        continue

    print("Waiting...")
    # Perhaps do something else here

p.disconnect()
```
which I complete its by open stackoverflow, finally I got

```
Waiting...
A notification was received: ��R��R��S �
A notification was received: ��R��R��S��
```

still not clear with ��R��R��S � 

It would be nice if you can update Docs , more example and usecase

PrzemoF, 2018-07-04:
About the notifications: it's all binary, like here [1]. You cannot just print it and expect something readable. Example of parsing it [2]. Another pair (spec & code) for speed & cadence [3] [4].
The examples might not work with python3 due to different handling of strings/bytes!

[1] https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.characteristic.heart_rate_measurement.xml
[2] https://github.com/PrzemoF/Open-Cycling-Computer/blob/master/code/src/ble_hr.py#L147
[3] https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.characteristic.csc_measurement.xml
[4] https://github.com/PrzemoF/Open-Cycling-Computer/blob/master/code/src/ble_sc.py#L164

avatarr, 2018-07-05:
Thank your for your answer very helpful especially the 3rd . Anyway, I've done with how to get a data already and it's in the phrase for design classes and application structure

closed 2018-07-05 by avatarr