IanHarvey/bluepy #289: memory leak seen while using getServices() call in a loop

SudhindraMewundi, 2018-08-06:
Hi Ian,

I am trying to connect to a BLE sensor device from RaspberryPi and get information by reading the characteristics on it. All works fine with the below skeleton code except that I see a small memory leak when used over a period of time. 
Below is the code that I use to connect and read from the characteristic. I am able to get the data. On my RaspberryPi I see that the memory increases by 0.1% for every 220+ counts of the while loop.

When checked for memory leak it was seen that only the lbleServlist = lPerHandle.getServices()     function was causing the leak and I am deleting the list below as del lbleServlist[:] since I am using pythin2.7 

Please suggest what might be the cause for the leak using the getServices() call on python 2.7.

Thanks
Sudhindra

def fble_connect(lMACaddr,lType,lThreadNum):
    global mutex
    
    while True:
    # start the BLE part here
        mutex.acquire()
        try:
            lPerHandle = btle.Peripheral(lMACaddr, lType)
            lbleServlist = lPerHandle.getServices()        #<----> Memory leak <---->
            lSensorService = lPerHandle.getServiceByUUID(btle.UUID("ae27a3a0-6ef8-46f2-a3ea-02d113fe1f2e"))
            lChArr = lSensorService.getCharacteristics()

            lCharacteristic = lChArr[0]
            lSensorArr = lCharacteristic.read()

            lPerHandle.disconnect()

            del lPerHandle
            del lChArr[:]
            del lbleServlist[:]
            del lSensorService
            del lSensorArr
            
            mutex.release()
            time.sleep(1)
        except:
            if DEBUG:
                print("Unable to connect yet....thread %d", lThreadNum)
            mutex.release()
            time.sleep(1)

Judas2016, 2019-01-31:
I am experiencing the same problem, reports back through Python as OS unable to allocate memory after long term testing cycles.

Jaftem, 2019-09-20:
Seeing the same, but also possibly for `getCharacteristics()`. Big issue when you want to scan keep your program running forever. Any updates?