IanHarvey/bluepy #291: Certain packets cause errors in `getScanData`

bunnymatic, 2018-08-08:
Using `bluepy` 1.2.0 with python 2.7.13, we're running basically the `blescan.py` script (with a few local modifications).

Every once in a while, we get a packet that fails to return it's scan data with the following exception

```
 File "/usr/local/lib/python2.7/dist-packages/bluepy/btle.py", line 656, in getScanData
    for sdid in self.scanData.keys() ]
  File "/usr/local/lib/python2.7/dist-packages/bluepy/btle.py", line 651, in getValueText
    return str(self.getValue(sdid))
UnicodeEncodeError: 'ascii' codec can't encode character u'\u2019' in position 7: ordinal not in range(128)
```

I don't have the exact packet data which I realize might make this hard to debug. But I wonder if there is some encoding trick that could be done in in the `getValue` method so that this would be a bit less error-prone.  I will likely add some `try/except` in my scanner code for the time being.

I'll be investigating in our system.  If I dig up more information, I'll try to add it here.

Any ideas what might be going on here?

bunnymatic, 2018-09-28:
Thanks for the update.  So moving to Python 3 might also be a good thing?

IanHarvey, 2018-12-03:
Yes... the Python maintainers are going to drop support for 2.x in 2020. Making bluepy a Python 3-only project will make it easier to maintain and release, so it's likely I'll stop officially supporting 2.x by 2020 as well.

closed 2018-12-03 by IanHarvey