IanHarvey/bluepy #292: UnicodeEncodeError Not Actually Fixed

lyokofirelyte, 2018-08-10:
I am running the latest version (compiled from GitHub) and am still getting this error on over 150 devices that this is deployed to.

[PiClient v1.4 > 08/10/2018 16:47:18] Traceback (most recent call last):
[PiClient v1.4 > 08/10/2018 16:47:18]   File "/usr/local/bin/blescan", line 11, in <module>
[PiClient v1.4 > 08/10/2018 16:47:18]     load_entry_point('bluepy==1.2.0', 'console_scripts', 'blescan')()
[PiClient v1.4 > 08/10/2018 16:47:18]   File "/usr/local/lib/python2.7/dist-packages/bluepy/blescan.py", line 122, in main
[PiClient v1.4 > 08/10/2018 16:47:18]     devices = scanner.scan(arg.timeout)
[PiClient v1.4 > 08/10/2018 16:47:18]   File "/usr/local/lib/python2.7/dist-packages/bluepy/btle.py", line 735, in scan
[PiClient v1.4 > 08/10/2018 16:47:18]     self.process(timeout)
[PiClient v1.4 > 08/10/2018 16:47:18]   File "/usr/local/lib/python2.7/dist-packages/bluepy/btle.py", line 724, in process
[PiClient v1.4 > 08/10/2018 16:47:18]     self.delegate.handleDiscovery(dev, (dev.updateCount <= 1), isNewData)
[PiClient v1.4 > 08/10/2018 16:47:18]   File "/usr/local/lib/python2.7/dist-packages/bluepy/blescan.py", line 89, in handleDiscovery
[PiClient v1.4 > 08/10/2018 16:47:18]     for (sdid, desc, val) in dev.getScanData():
[PiClient v1.4 > 08/10/2018 16:47:18]   File "/usr/local/lib/python2.7/dist-packages/bluepy/btle.py", line 656, in getScanData
[PiClient v1.4 > 08/10/2018 16:47:18]     for sdid in self.scanData.keys() ]
[PiClient v1.4 > 08/10/2018 16:47:18]   File "/usr/local/lib/python2.7/dist-packages/bluepy/btle.py", line 651, in getValueText
[PiClient v1.4 > 08/10/2018 16:47:18]     return str(self.getValue(sdid))
[PiClient v1.4 > 08/10/2018 16:47:18] UnicodeEncodeError: 'ascii' codec can't encode character u'\xed' in position 1: ordinal not in range(128)

closed 2018-09-24 by IanHarvey

IanHarvey, 2018-09-24:
I've just pushed a633e489 which fixes `getValueText()` to return the same as `getValue()` for names. It would be useful to know if this fixes the problem for your devices.

lyokofirelyte, 2018-09-24:
Thanks for the update. I've pushed the update to one of the live devices
that was having the error and I'll let you know if I see the error again.
So far so good.

lyokofirelyte, 2018-12-20:
Hi, getting the problem again on version 1.3.0, any ideas?
[image: image.png]

[image: image.png]

commit a633e489 (2018-09-24): Improve ScanEntry.getValue() and .getValueText(), fixes #292.

getValueText() now returns 'str' or 'unicode' for names.
UTF-8 decoding errors are returned as strings with '?' for unknown
characters.

Note that getValue() now returns byteblocks, not ASCII hex, for
unknown AD type items. getValueText() continues to return hex.