IanHarvey/bluepy #293: running install Failed to compile bluepy-helper. Exiting install.

digitalgreentea, 2018-08-10:
When I run the command `python setup.py install` after building `python setup.py build` it outputs the following: 

```
running install
Failed to compile bluepy-helper. Exiting install.
Command was 'make -C ./bluepy' in /cygdrive/c/Program Files/OpenBCI_Python/OpenB  CI_Python/bluepy
Return code was 2
Output was:
make: Entering directory '/cygdrive/c/Program Files/OpenBCI_Python/OpenBCI_Pytho  n/bluepy/bluepy'
tar xzf bluez-src.tgz
touch ./bluez-5.29/lib/bluetooth.c ./bluez-5.29/lib/hci.c ./bluez-5.29/lib/sdp.c   ./bluez-5.29/lib/uuid.c ./bluez-5.29/attrib/att.c ./bluez-5.29/attrib/gatt.c ./  bluez-5.29/attrib/gattrib.c ./bluez-5.29/attrib/utils.c ./bluez-5.29/btio/btio.c   ./bluez-5.29/src/log.c ./bluez-5.29/src/shared/mgmt.c ./bluez-5.29/src/shared/c  rypto.c ./bluez-5.29/src/shared/att.c ./bluez-5.29/src/shared/queue.c ./bluez-5.  29/src/shared/util.c ./bluez-5.29/src/shared/io-glib.c ./bluez-5.29/src/shared/t  imeout-glib.c
gcc -L. -Os -g -Wall -Werror -DHAVE_CONFIG_H -I./bluez-5.29/attrib -I./bluez-5.2  9 -I./bluez-5.29/lib -I./bluez-5.29/src -I./bluez-5.29/gdbus -I./bluez-5.29/btio   -I/usr/include/glib-2.0 -I/usr/lib/glib-2.0/include -o bluepy-helper bluepy-hel  per.c ./bluez-5.29/lib/bluetooth.c ./bluez-5.29/lib/hci.c ./bluez-5.29/lib/sdp.c   ./bluez-5.29/lib/uuid.c ./bluez-5.29/attrib/att.c ./bluez-5.29/attrib/gatt.c ./  bluez-5.29/attrib/gattrib.c ./bluez-5.29/attrib/utils.c ./bluez-5.29/btio/btio.c   ./bluez-5.29/src/log.c ./bluez-5.29/src/shared/mgmt.c ./bluez-5.29/src/shared/c  rypto.c ./bluez-5.29/src/shared/att.c ./bluez-5.29/src/shared/queue.c ./bluez-5.  29/src/shared/util.c ./bluez-5.29/src/shared/io-glib.c ./bluez-5.29/src/shared/t  imeout-glib.c -lglib-2.0 -lintl
./bluez-5.29/btio/btio.c: In function ‘bt_io_get_type’:
./bluez-5.29/btio/btio.c:111:35: error: ‘SO_DOMAIN’ undeclared (first use in thi  s function); did you mean ‘G_LOG_DOMAIN’?
  err = getsockopt(sk, SOL_SOCKET, SO_DOMAIN, &domain, &len);
                                   ^~~~~~~~~
                                   G_LOG_DOMAIN
./bluez-5.29/btio/btio.c:111:35: note: each undeclared identifier is reported on  ly once for each function it appears in
./bluez-5.29/btio/btio.c:125:35: error: ‘SO_PROTOCOL’ undeclared (first use in t  his function)
  err = getsockopt(sk, SOL_SOCKET, SO_PROTOCOL, &proto, &len);
                                   ^~~~~~~~~~~
./bluez-5.29/btio/btio.c: In function ‘set_priority’:
./bluez-5.29/btio/btio.c:573:35: error: ‘SO_PRIORITY’ undeclared (first use in t  his function); did you mean ‘SO_ERROR’?
  if (setsockopt(sock, SOL_SOCKET, SO_PRIORITY, &prio, sizeof(prio)) < 0)
                                   ^~~~~~~~~~~
                                   SO_ERROR
./bluez-5.29/btio/btio.c: In function ‘get_priority’:
./bluez-5.29/btio/btio.c:965:35: error: ‘SO_PRIORITY’ undeclared (first use in t  his function); did you mean ‘SO_ERROR’?
  if (getsockopt(sock, SOL_SOCKET, SO_PRIORITY, prio, &len) < 0)
                                   ^~~~~~~~~~~
                                   SO_ERROR
./bluez-5.29/src/shared/att.c: In function ‘is_io_l2cap_based’:
./bluez-5.29/src/shared/att.c:867:35: error: ‘SO_DOMAIN’ undeclared (first use i  n this function)
  err = getsockopt(fd, SOL_SOCKET, SO_DOMAIN, &domain, &len);
                                   ^~~~~~~~~
./bluez-5.29/src/shared/att.c:867:35: note: each undeclared identifier is report  ed only once for each function it appears in
./bluez-5.29/src/shared/att.c:876:35: error: ‘SO_PROTOCOL’ undeclared (first use   in this function)
  err = getsockopt(fd, SOL_SOCKET, SO_PROTOCOL, &proto, &len);
                                   ^~~~~~~~~~~
make: *** [Makefile:28: bluepy-helper] Error 1
make: Leaving directory '/cygdrive/c/Program Files/OpenBCI_Python/OpenBCI_Python  /bluepy/bluepy'
```

ycathi, 2018-11-11:
I have the same problem. 

My device information：

Updatable packages:
All packages up to date
System information:
Linux localhost 3.10.84-perf-g7f0277a #1 SMP PREEMPT Thu Nov 8 00:40:04 CST 2018 aarch64 Android
Termux-packages arch:
aarch64
Android version:
6.0.1
Device manufacturer:
Xiaomi
Device model:
Redmi Note 3

ycathi, 2018-11-11:
Collecting bluepy
  Using cached https://files.pythonhosted.org/packages/1f/5d/fb28bf7d95669b6909cc39dbc1c631f16fc61184f4bb5a3188f1662ca59c/bluepy-1.2.0.tar.gz
Installing collected packages: bluepy
  Running setup.py install for bluepy: started
    Running setup.py install for bluepy: finished with status 'error'
    Complete output from command /data/data/com.termux/files/usr/bin/python -u -c "import setuptools, tokenize;__file__='/data/data/com.termux/files/usr/tmp/pip-install-b3b94it2/bluepy/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /data/data/com.termux/files/usr/tmp/pip-record-1acdczca/install-record.txt --single-version-externally-managed --compile:
    running install
    running build
    running build_py
    Working dir is /data/data/com.termux/files/usr/tmp/pip-install-b3b94it2/bluepy
    execute make -C ./bluepy clean
    execute make -C bluepy -j1
    Failed to compile bluepy-helper. Exiting install.
    Command was 'make -C bluepy -j1' in /data/data/com.termux/files/usr/tmp/pip-install-b3b94it2/bluepy
    Return code was 2
    Output was:
    b"make: Entering directory '/data/data/com.termux/files/usr/tmp/pip-install-b3b94it2/bluepy/bluepy'\ntar xzf bluez-src.tgz\ntouch ./bluez-5.47/lib/bluetooth.c ./bluez-5.47/lib/hci.c ./bluez-5.47/lib/sdp.c ./bluez-5.47/lib/uuid.c ./bluez-5.47/attrib/att.c ./bluez-5.47/attrib/gatt.c ./bluez-5.47/attrib/gattrib.c ./bluez-5.47/attrib/utils.c ./bluez-5.47/btio/btio.c ./bluez-5.47/src/log.c ./bluez-5.47/src/shared/mgmt.c ./bluez-5.47/src/shared/crypto.c ./bluez-5.47/src/shared/att.c ./bluez-5.47/src/shared/queue.c ./bluez-5.47/src/shared/util.c ./bluez-5.47/src/shared/io-glib.c ./bluez-5.47/src/shared/timeout-glib.c\ncc -L. -Os -g -Wall  -DHAVE_CONFIG_H -I./bluez-5.47/attrib -I./bluez-5.47 -I./bluez-5.47/lib -I./bluez-5.47/src -I./bluez-5.47/gdbus -I./bluez-5.47/btio -I./bluez-5.47/sys -I/data/data/com.termux/files/usr/include/glib-2.0 -I/data/data/com.termux/files/usr/lib/glib-2.0/include -I/data/data/com.termux/files/usr/include -o bluepy-helper bluepy-helper.c ./bluez-5.47/lib/bluetooth.c ./bluez-5.47/lib/hci.c ./bluez-5.47/lib/sdp.c ./bluez-5.47/lib/uuid.c ./bluez-5.47/attrib/att.c ./bluez-5.47/attrib/gatt.c ./bluez-5.47/attrib/gattrib.c ./bluez-5.47/attrib/utils.c ./bluez-5.47/btio/btio.c ./bluez-5.47/src/log.c ./bluez-5.47/src/shared/mgmt.c ./bluez-5.47/src/shared/crypto.c ./bluez-5.47/src/shared/att.c ./bluez-5.47/src/shared/queue.c ./bluez-5.47/src/shared/util.c ./bluez-5.47/src/shared/io-glib.c ./bluez-5.47/src/shared/timeout-glib.c -L/data/data/com.termux/files/usr/lib -lglib-2.0\n/data/data/com.termux/files/usr/bin/aarch64-linux-android-ld: /data/data/com.termux/files/usr/tmp/sdp-912f57.o: in function `android_polyfill_vsyslog':\n/data/data/com.termux/files/usr/include/syslog.h:119: undefined reference to `__android_log_print'\n/data/data/com.termux/files/usr/bin/aarch64-linux-android-ld: /data/data/com.termux/files/usr/include/syslog.h:116: undefined reference to `__android_log_vprint'\n/data/data/com.termux/files/usr/bin/aarch64-linux-android-ld: /data/data/com.termux/files/usr/tmp/log-8c2b31.o: in function `android_polyfill_vsyslog':\n/data/data/com.termux/files/usr/include/syslog.h:119: undefined reference to `__android_log_print'\n/data/data/com.termux/files/usr/bin/aarch64-linux-android-ld: /data/data/com.termux/files/usr/include/syslog.h:116: undefined reference to `__android_log_vprint'\n/data/data/com.termux/files/usr/bin/aarch64-linux-android-ld: /data/data/com.termux/files/usr/include/syslog.h:119: undefined reference to `__android_log_print'\n/data/data/com.termux/files/usr/bin/aarch64-linux-android-ld: /data/data/com.termux/files/usr/include/syslog.h:116: undefined reference to `__android_log_vprint'\n/data/data/com.termux/files/usr/bin/aarch64-linux-android-ld: /data/data/com.termux/files/usr/include/syslog.h:119: undefined reference to `__android_log_print'\n/data/data/com.termux/files/usr/bin/aarch64-linux-android-ld: /data/data/com.termux/files/usr/include/syslog.h:116: undefined reference to `__android_log_vprint'\n/data/data/com.termux/files/usr/bin/aarch64-linux-android-ld: /data/data/com.termux/files/usr/include/syslog.h:119: undefined reference to `__android_log_print'\n/data/data/com.termux/files/usr/bin/aarch64-linux-android-ld: /data/data/com.termux/files/usr/include/syslog.h:116: undefined reference to `__android_log_vprint'\n/data/data/com.termux/files/usr/bin/aarch64-linux-android-ld: /data/data/com.termux/files/usr/include/syslog.h:119: undefined reference to `__android_log_print'\n/data/data/com.termux/files/usr/bin/aarch64-linux-android-ld: /data/data/com.termux/files/usr/include/syslog.h:116: undefined reference to `__android_log_vprint'\n/data/data/com.termux/files/usr/bin/aarch64-linux-android-ld: /data/data/com.termux/files/usr/include/syslog.h:119: undefined reference to `__android_log_print'\n/data/data/com.termux/files/usr/bin/aarch64-linux-android-ld: /data/data/com.termux/files/usr/include/syslog.h:116: undefined reference to `__android_log_vprint'\n/data/data/com.termux/files/usr/bin/aarch64-linux-android-ld: /data/data/com.termux/files/usr/include/syslog.h:119: undefined reference to `__android_log_print'\n/data/data/com.termux/files/usr/bin/aarch64-linux-android-ld: /data/data/com.termux/files/usr/include/syslog.h:116: undefined reference to `__android_log_vprint'\n/data/data/com.termux/ [... cut ...] log.h:119: undefined reference to `__android_log_print'\n/data/data/com.termux/files/usr/bin/aarch64-linux-android-ld: /data/data/com.termux/files/usr/include/syslog.h:116: undefined reference to `__android_log_vprint'\nclang-7: error: linker command failed with exit code 1 (use -v to see invocation)\nmake: *** [Makefile:28: bluepy-helper] Error 1\nmake: Leaving directory '/data/data/com.termux/files/usr/tmp/pip-install-b3b94it2/bluepy/bluepy'\n"
    
    ----------------------------------------