IanHarvey/bluepy #296: double free or corruption (fasttop): 0x000055d584d82580

eknoes, 2018-08-16:
Hey, when I use [Bleah](https://github.com/evilsocket/bleah/) to write certain data to a specific handle, I get the following error:

```
*** Error in `/usr/local/lib/python2.7/dist-packages/bluepy-1.1.4-py2.7.egg/bluepy/bluepy-helper': double free or corruption (fasttop): 0x000055d584d82580 ***
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x70bfb)[0x7f197032dbfb]
/lib/x86_64-linux-gnu/libc.so.6(+0x76fc6)[0x7f1970333fc6]
/lib/x86_64-linux-gnu/libc.so.6(+0x7780e)[0x7f197033480e]
/usr/local/lib/python2.7/dist-packages/bluepy-1.1.4-py2.7.egg/bluepy/bluepy-helper(+0x1651a)[0x55d5845c551a]
/usr/local/lib/python2.7/dist-packages/bluepy-1.1.4-py2.7.egg/bluepy/bluepy-helper(+0x1d3ae)[0x55d5845cc3ae]
/usr/local/lib/python2.7/dist-packages/bluepy-1.1.4-py2.7.egg/bluepy/bluepy-helper(+0x1d3e3)[0x55d5845cc3e3]
/usr/local/lib/python2.7/dist-packages/bluepy-1.1.4-py2.7.egg/bluepy/bluepy-helper(+0x16874)[0x55d5845c5874]
/usr/local/lib/python2.7/dist-packages/bluepy-1.1.4-py2.7.egg/bluepy/bluepy-helper(+0x7804)[0x55d5845b6804]
/usr/local/lib/python2.7/dist-packages/bluepy-1.1.4-py2.7.egg/bluepy/bluepy-helper(+0x664d)[0x55d5845b564d]
/lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_dispatch+0x15a)[0x7f19706a66aa]
/lib/x86_64-linux-gnu/libglib-2.0.so.0(+0x4aa60)[0x7f19706a6a60]
/lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_loop_run+0xc2)[0x7f19706a6d82]
/usr/local/lib/python2.7/dist-packages/bluepy-1.1.4-py2.7.egg/bluepy/bluepy-helper(+0x5ae2)[0x55d5845b4ae2]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf1)[0x7f19702dd2e1]
/usr/local/lib/python2.7/dist-packages/bluepy-1.1.4-py2.7.egg/bluepy/bluepy-helper(+0x5baa)[0x55d5845b4baa]
======= Memory map: ========
55d5845af000-55d5845de000 r-xp 00000000 08:01 1067867                    /usr/local/lib/python2.7/dist-packages/bluepy-1.1.4-py2.7.egg/bluepy/bluepy-helper
55d5847dd000-55d5847de000 r--p 0002e000 08:01 1067867                    /usr/local/lib/python2.7/dist-packages/bluepy-1.1.4-py2.7.egg/bluepy/bluepy-helper
55d5847de000-55d5847e1000 rw-p 0002f000 08:01 1067867                    /usr/local/lib/python2.7/dist-packages/bluepy-1.1.4-py2.7.egg/bluepy/bluepy-helper
55d584d78000-55d584d99000 rw-p 00000000 00:00 0                          [heap]
7f1968000000-7f1968021000 rw-p 00000000 00:00 0 
7f1968021000-7f196c000000 ---p 00000000 00:00 0 
7f196fc16000-7f196fc2c000 r-xp 00000000 08:01 262148                     /lib/x86_64-linux-gnu/libgcc_s.so.1
7f196fc2c000-7f196fe2b000 ---p 00016000 08:01 262148                     /lib/x86_64-linux-gnu/libgcc_s.so.1
7f196fe2b000-7f196fe2c000 r--p 00015000 08:01 262148                     /lib/x86_64-linux-gnu/libgcc_s.so.1
7f196fe2c000-7f196fe2d000 rw-p 00016000 08:01 262148                     /lib/x86_64-linux-gnu/libgcc_s.so.1
7f196fe2d000-7f196fe45000 r-xp 00000000 08:01 262167                     /lib/x86_64-linux-gnu/libpthread-2.24.so
7f196fe45000-7f1970044000 ---p 00018000 08:01 262167                     /lib/x86_64-linux-gnu/libpthread-2.24.so
7f1970044000-7f1970045000 r--p 00017000 08:01 262167                     /lib/x86_64-linux-gnu/libpthread-2.24.so
7f1970045000-7f1970046000 rw-p 00018000 08:01 262167                     /lib/x86_64-linux-gnu/libpthread-2.24.so
7f1970046000-7f197004a000 rw-p 00000000 00:00 0 
7f197004a000-7f19700bc000 r-xp 00000000 08:01 262183                     /lib/x86_64-linux-gnu/libpcre.so.3.13.3
7f19700bc000-7f19702bb000 ---p 00072000 08:01 262183                     /lib/x86_64-linux-gnu/libpcre.so.3.13.3
7f19702bb000-7f19702bc000 r--p 00071000 08:01 262183                     /lib/x86_64-linux-gnu/libpcre.so.3.13.3
7f19702bc000-7f19702bd000 rw-p 00072000 08:01 262183                     /lib/x86_64-linux-gnu/libpcre.so.3.13.3
7f19702bd000-7f1970452000 r-xp 00000000 08:01 262152                     /lib/x86_64-linux-gnu/libc-2.24.so
7f1970452000-7f1970652000 ---p 00195000 08:01 262152                     /lib/x86_64-linux-gnu/libc-2.24.so
7f1970652000-7f1970656000 r--p 00195000 08:01 262152                     /lib/x86_64-linux-gnu/libc-2.24.so
7f1970656000-7f1970658000 rw-p 00199000 08:01 262152                     /lib/x86_64-linux-gnu/libc-2.24.so
7f1970658000-7f197065c000 rw-p 00000000 00:00 0 
7f197065c000-7f197076e000 r-xp 00000000 08:01 266232                     /lib/x86_64-linux-gnu/libglib-2.0.so.0.5000.3
7f197076e000-7f197096d000 ---p 00112000 08:01 266232                     /lib/x86_64-linux-gnu/libglib-2.0.so.0.5000.3
7f197096d000-7f197096e000 r--p 00111000 08:01 266232                     /lib/x86_64-linux-gnu/libglib-2.0.so.0.5000.3
7f197096e000-7f197096f000 rw-p 00112000 08:01 266232                     /lib/x86_64-linux-gnu/libglib-2.0.so.0.5000.3
7f197096f000-7f1970970000 rw-p 00000000 00:00 0 
7f1970970000-7f1970993000 r-xp 00000000 08:01 262147                     /lib/x86_64-linux-gnu/ld-2.24.so
7f1970b72000-7f1970b74000 rw-p 00000000 00:00 0 
7f1970b8f000-7f1970b93000 rw-p 00000000 00:00 0 
7f1970b93000-7f1970b94000 r--p 00023000 08:01 262147                     /lib/x86_64-linux-gnu/ld-2.24.so
7f1970b94000-7f1970b95000 rw-p 00024000 08:01 262147                     /lib/x86_64-linux-gnu/ld-2.24.so
7f1970b95000-7f1970b96000 rw-p 00000000 00:00 0 
7ffe23f3e000-7ffe23f5f000 rw-p 00000000 00:00 0                          [stack]
7ffe23f76000-7ffe23f78000 r--p 00000000 00:00 0                          [vvar]
7ffe23f78000-7ffe23f7a000 r-xp 00000000 00:00 0                          [vdso]
ffffffffff600000-ffffffffff601000 r-xp 00000000 00:00 0                  [vsyscall]
```

dmesg:
```
[15112.634694] bluepy-helper[8931]: segfault at 55f70000000b ip 00007f331e913248 sp 00007ffe11bbdd30 error 4 in libc-2.24.so[7f331e89c000+195000]
```

As this error appears to me using bleah, this is my call:
```
bleah -b "30:AE:A4:6A:BC:9A" -n 0x0050 -d "34028236692093846346"
```

If you tell me how to debug this further, I am happy doing so!

afcady, 2019-01-12:
I've got a similar looking segfault, but it happens unpredictably.  I have some code that scans for BLE devices, waiting for a certain match, then lists the characteristics of the matching device.  It runs in a loop forever.  Except it never does run forever, because of this segfault.  It seems to go about an hour (during which my two devices come alive for 12 times each, for 12 seconds at a time, producing 4 beacons per second, so ~1200 beacons processed and about ~20 characteristics listings produced -- because it only produces one such listing each on period per device -- before crash).

I guess `bluepy-helper` ought to be run in `gdb`.  For me I don't think I can reproduce this enough to consider debugging it yet (actually I've been working on moving away from using this library entirely) but if you're willing to debug, then I will try to help.

So far I have managed to get it running in `gdb`.  What I did was this:

* git clone https://github.com/IanHarvey/bluepy/
* cd bluepy
* sed -i -e '1i DEBUGGING=y' bluepy/Makefile
* python3 setup.py build
* python3 setup.py install
* [restart my program]
* gdb -p $(pidof bluepy-helper)
* [type `continue` in gdb]

I literally just did this, and the program hasn't crashed yet since I did, so I have no gdb results.

That approach won't work if your `bluepy-helper` is crashing close to startup, not giving `gdb -p` enough time to attach.  For that you might be able to use some kind of wrapper.  I see that `bluepy-helper` is called here: https://github.com/IanHarvey/bluepy/blob/53ce2f2388a936663b94f00636cc2e3677325182/bluepy/btle.py#L22 .  It is just using $PATH to find the executable.  So you could put a wrapper of the same name in the front of the $PATH, and have it call the real `bluepy-helper` via `gdb`.  I am not sure exactly how to do that though.  But it's probably been done before.

----

OK, I just discovered that `bluepy-helper` actually gets restarted by `bluepy`.  So `gdb -p` isn't really sufficient.  It needs to be re-attached time and time again.  Just going to post this message and do something else for a while now.

afcady, 2019-01-12:
https://github.com/yugr/libdebugme