IanHarvey/bluepy #303: Decreasing "Try to Connect" timeout

kesslera, 2018-09-12:
Hi Ian, 

I am currently trying to build a sensor network with a Raspi 3b collecting the data via Bluetooth and pushing it to the cloud. The problem is, that sometimes the raspi is unable to connect to a sensor and runs into a timeout error "BLEError unable to connect to peripheral", which is fine. But the time it takes to run into the timeout is too long for my approach as it is stopping the whole process of scanning all sensor and connecting to them afterwards. 

I haven't found a parameter in the documentation, but in the source code of btle.py there is a timeout parameter in the _waitResp function. Can I use this timeout? Or is there another way to force quit the connection-process after some seconds without success ? 

Thanks in advance,
Alex

PrzemoF, 2018-09-17:
Take a look here: https://stackoverflow.com/questions/28298980/how-can-i-set-the-ble-connection-timeout-of-gatttool-on-raspberry-pi#28397728
The timeout is not defined in bluepy as far I I can tell. Alternative way:
1. Start connection in a separate thread
2. Wait as long as you need
3. Kill the thread in not connected
Not very elegant way, but it should work.

MatteoBaldini, 2018-09-22:
@PrzemoF Your solution unfortunately won't work. If the connection is started on a separate thread the original thread still needs to use bluetooth resources to interact with other devices and this can't be achieved if there's only one bluetooth chip (which I'm assuming is the case). I have a similar isue and to solve it I'm changing my approach completely.

kesslera, 2018-09-22:
I've now a hardcoded timeout solution/workaround for this:

```Python
    def _getResp(self, wantType, timeout=3):
    [...]

    def _connect(self, addr, addrType=ADDR_TYPE_PUBLIC, iface=None):
    [...]
        if rsp == None:
            raise BTLEException(BTLEException.DISCONNECTED,
                                "Failed to connect to peripheral %s, addr type: %s" % (addr, addrType))
        while rsp['state'][0] == 'tryconn':
            rsp = self._getResp('stat')
            if rsp == None:
                raise BTLEException(BTLEException.DISCONNECTED,
                                    "Failed to connect to peripheral %s, addr type: %s" % (addr, addrType))
        if rsp['state'][0] != 'conn':
            self._stopHelper()
            raise BTLEException(BTLEException.DISCONNECTED,
                                "Failed to connect to peripheral %s, addr type: %s" % (addr, addrType))
```