IanHarvey/bluepy #307: ScanEntry

MatteoBaldini, 2018-09-21:
Hello

I'm trying to get some data from a ScanEntry object. From the docs it's clear that using the getScanData method will return a list of tuples with adtype, description and value. This all works fine but the data gathered is quite obscure to me. For example:
adtype | description | value
1 | Flags | 1a
255 | Manufacturer | 4c001005011c72a858
255 | Manufacturer | 4c0009060309c0a8122e
255 | Manufacturer | 060001092002a58a31e2f44e4f587a306ba07a643f50570b53baa242b2
255 | Manufacturer | 4c0010050b109edbbf

What do these number mean? I've looked at the Bluetooth website and all they list are hex values so what are these? How can I turn these into usable information (Meaning how do I get the actual manufacturer instead of the code? Is there a database of manufacturer codes?

Thanks in advance and apologies if it's a stupid question

mikimer, 2018-09-21:
I've had to contact some manufacturers to get documentation on how to decode their hex payloads.
Sometimes you can just convert the hex to ASCII (https://www.rapidtables.com/convert/number/hex-to-ascii.html) -- I tried that with your examples, but didn't come up with anything. 
The best thing I've found is a hex payload that translated to "😎 Specs"

MatteoBaldini, 2018-09-22:
Thanks a lot! I'm new to this technology so it's a bit confusing. @ukBaz  The article mentions a 3rd part of the device name. What about the rest of the bytes? Could you elaborate on it? because the examples they show don't match the type of payload I'm getting.

@mikimer Fair enough, I've tried many ways to convert the payload but with zero luck. The answer above yours partially explains my questions but what about the rest of the payload?

Thanks again

ukBaz, 2018-09-22:
@MatteoBaldini the manufacturer specific information is specific to that manufacturer and they may not make the format publicly available. 
Do you know what those devices are?
If they are ibeacons, that information is available. If you look at the following article there is a chart by Jos comparing the different beacon formats
https://ukbaz.github.io/howto/beacon_scan_cmd_line.html