IanHarvey/bluepy #310: Missing advertising packets

mikimer, 2018-09-26:
I'm using bluepy on a Raspberry Pi to scan for BLE packets and I've discovered that the RasPi is missing most of the packets. Of course, I want to capture as many packets as possible, so I wanted to ask for help...

**My test**
I have 16 beacons in a room broadcasting BLE advertising packets at a rate of 1 packet per second.  I gathered data for all the beacons using two RasPi's and analyzed the results for 1 of the 16 beacons in a histogram (attached). Of the beacon's packets that the RasPi's captured, only 30% to 40% arrived within 1 second of the previous packet, indicating no loss. The rest of the packets arrived 2 to 9 seconds after the previous packet, meaning that the RasPi missed the beacon's packets in the interim. 

I validated that the problem is with the RasPi's and not the beacon by comparing results between the two RasPi's. I positioned the RasPi's side by side, so they would receive the same packets. The beacon I'm examining is about 12 feet from the RasPi's.  If the beacon didn't perform as advertised and instead sent a packet every 1-10 seconds, then I would have expected that the dataset from RasPi_1 would be a duplicate of the dataset from RasPi_2. However, I analyzed the datasets and found that only 20% of the packets were duplicates; 80% of the packets were unique to each RasPi, meaning each RasPi caught a bunch of packets that the other RasPi missed.

**My questions**
Can anyone offer advice on what's causing this problem and how to fix it? 
Is this problem due to the RasPi? the bluepy library? 
Should I use a different library, like [beacontools](https://github.com/citruz/beacontools)? 
Would it help if I put multiple BLE dongles on my RasPi? 
Is the problem due to BLE packets from the other 9 beacons "colliding" with the packets from the 1 beacon I'm examining? i.e., Is there an upper limit to how many BLE packets a RasPi can read at once? 

[image]

gandy92, 2018-09-27:
Do you perform active or passive scanning? I've experienced similar issues with active scanning, while passive scanning misses close to none of the advertising packets.

mikimer, 2018-09-28:
@gandy92 thanks for the tip! I'll see if that fixes the problem.