IanHarvey/bluepy #315: Version 1.2 disconnects more often

EinSoldiatGott, 2018-10-05:
Hi, I was using version 1.4.4 and then I realized that a new version of bluepy was realesed 1.2

I updated but then I started to notice a lot of disconnection issues, Exception: something like disconnected from Mac... type random.

I returned to 1.4.4 and it's stable again, my issue has a lack of information but if more users starts to report this problem you could take a look at what happened.

I'll try to install versions between them to find which one is the problem.

jajosheni, 2019-12-15:
is this solved or improved in any new version?

jajosheni, 2019-12-18:
I think I kinda know what is going on... I tried connecting an external BLE dongle and it works like charm.
With the inboard BLE chip even gatttool disconnects a lot.

So buy a cheap 5$ dongle and attach it to your rpi.