IanHarvey/bluepy #320: Raspberry Pi can not connect to nrf52

hb9608, 2018-10-17:
I am using 4.9.59 kernel and 5.43 bluez version on rpi 3. I am trying to connect to nrf52. I have successfully connected to nrf52 by following 
http://infocenter.nordicsemi.com/topic/com.nordic.infocenter.sdk5.v15.2.0/iot_ug_border_router.html
Here i used command line interface but i'd like to do that with python with bluepy. Scanning with bluepy is okey but connecting to nrf52 does not yield the result. The code i used in the following:

`
from bluepy import btle

dev=btle.Peripheral("00:73:C6:D5:49:A9")
lightSensor = btle.UUID("00001820-0000-1000-8000-00805f9b34fb")
lightService = dev.getServiceByUUID(lightSensor)
for ch in lightService.getCharacteristics():
     print str(ch)
`
When i observe hcidump, the observations like that:

[image]

How can i solve this issue, what is the way that it is always connected?

steveww, 2018-11-12:
Use bale.Peripheral(address, btle.ADDR_TYPE_RANDOM)
Works for me but no idea why.

dhruvkakadiya, 2019-03-03:
@hb9608 use `dev=btle.Peripheral("00:73:C6:D5:49:A9", btle.ADDR_TYPE_RANDOM)` because default address type is `btle.ADDR_TYPE_PUBLIC` when you don't pass any argument while nRF52 peripheral expects device type as `btle.ADDR_TYPE_RANDOM`.