IanHarvey/bluepy #345: Send HEX

nanoymaster, 2019-02-12:
pulling my hair out here.

is it possible to send hex to a characteristic?
e.g. using gatttool:
╰» gatttool -b 24:0a:c4:9a:7b:8e --char-read -a 0x0036|awk -F':' '{print $2}'|tr -d ' '|xxd -r -p;printf '\n' 
Write the hex value 0x07 here
 
╰» gatttool -b 24:0a:c4:9a:7b:8e --char-write-req -a 0x0036 -n 07
Characteristic value was written successfully

that works (-n 07)
however I can't work out how to do this using bluepy:

```
x = 7
#newhex = '0x{:02x}'.format(x)
#grr = hex(x)
#print (grr)
#xint = int(x)
#xchr = chr(xint)
#xhex = '{:02x}'.format(xint)
xlify = binascii.hexlify(bytes([7]))
#sendhex = bytes([xint])
#print (xhex)
#p.disconnect()
#p = btle.Peripheral("24:0a:c4:9a:7b:8e")
#p.getServices()
#service=p.getServiceByUUID("000000ff-0000-1000-8000-00805f9b34fb")
#print ("Sending: "+sendhex)
#sendhex = bytes([x])
val  = struct.pack(">B", 0x07)
#bhex = bytes.fromhex(x)
print(val)
ch = p.getCharacteristics(chall_uuid)[0]
#p.writeCharacteristic(ch.valHandle, xlify)
response = ch.write(val , withResponse = True)
ch = p.getCharacteristics(chall_uuid)[0]
#response = ch.write(struct.pack('<B', x));
#test = struct.pack('<B', x)
#print test
#response = ch.write(sendhex, withResponse = True)
#print (response) 
#ch.write(sendhex, True)
hex2 = binascii.b2a_hex(p.readCharacteristic(chall_uuid)) 
hexlif2 = binascii.unhexlify(hex2)
print(hexlif2)
hexstr2 = str(hex, 'utf-8')
if hexstr2 != hexstr:
    print (hexstr2) 
    hexlif2 = binascii.unhexlify(hex2) # second step in tutorial
    hexlif2 = str(hexlif2, 'utf-8')
    print ("Unhexlified: " + hexlif2)
```

clearly I don't really know python that well!

closed 2019-02-13 by nanoymaster

parkerlreed, 2019-02-13:
Please don't edit your post. Other people may have the same issue. Now they have no clue what the issue was or the solution.

https://xkcd.com/979/

nanoymaster, 2019-02-13:
oh, ok sure...
so problem was I was struggling to send hex (0x07) to a characteristic.

Heres how to do do it:
  
```
#! /usr/bin/python3
import bluepy.btle as btle

chall_uuid = 0x36

p = btle.Peripheral("24:0a:c4:9a:7b:8e")
print ("Attached to peripheral")
services=p.getServices()
for service in services:
   print (service)
svc=p.getServiceByUUID(0x00FF)
print(svc)

# the number to turn into hex (0x07)
x = 7 
# put into bytearrray for python3
sendhex = bytes([x]) 
print (sendhex) 

# send to the device, true=withresponse
response = p.writeCharacteristic(chall_uuid, sendhex, True)
print (response) 
```

Solution found from: https://github.com/IanHarvey/bluepy/issues/53

Hope this helps someone